betaflight/betaflight #3652: 15x15mm Eachine TeenyCube FC.  Any no hope of telemetry?

derekmoc, 2017-07-28:
Hi all,
Foolishly assumed that all FCs came with a smartport option these days and bought one of these

https://www.banggood.com/15x15mm-Eachine-TeenyCube-Spare-Part-F3-6Dof-Betaflight-2-4S-Flight-Control-with-5V-BEC-p-1135796.html?rmmds=search

Just realized there is no mention of telemetry or smartport anywhere.  Its a long shot but can I use the Led Strip somehow by remapping resources?

jflyper, 2017-07-28:
Do this
```
resource serial_tx 11 a8
feature softserial
feature telemetry
serial 30 32 115200 57600 0 115200
save
```

derekmoc, 2017-07-29:
Thanks jflyper,

No luck with your suggestion, tried "resource serial_tx 11  A08" as well as "resource serial_tx 11 a8" as the led_strip resource used A08.  Dump of _pre_ your changes below.

Receiver is the XSR-M and I have tested it on a different FC (EMAX Femto) and battery cell voltage comes through fine on A4 as well as vfras and all the other usual things.

2017-07-29 @ 00:21:44 -- Running - OS: Windows, Chrome: 59.0.3071.115, Configurator: 3.1.1
2017-07-29 @ 00:21:47 -- Serial port successfully opened with ID: 1
2017-07-29 @ 00:21:47 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:21:48 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:21:49 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:21:50 -- Board: OMNI, version: 0
2017-07-29 @ 00:21:51 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:22:22 -- Serial port successfully closed
2017-07-29 @ 00:46:31 -- Serial port successfully opened with ID: 2
2017-07-29 @ 00:46:41 -- No configuration received within 10 seconds, communication failed
2017-07-29 @ 00:46:41 -- Serial port successfully closed
2017-07-29 @ 00:46:49 -- Serial port successfully opened with ID: 3
2017-07-29 @ 00:46:50 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:46:52 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:46:52 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:46:52 -- Board: OMNI, version: 0
2017-07-29 @ 00:46:53 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:47:19 -- EEPROM saved
2017-07-29 @ 00:47:19 -- Device - Rebooting
2017-07-29 @ 00:47:19 -- Serial port successfully closed
2017-07-29 @ 00:47:21 -- Serial port successfully opened with ID: 4
2017-07-29 @ 00:47:21 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:47:21 -- Serial port successfully closed
2017-07-29 @ 00:47:23 -- Serial port successfully opened with ID: 5
2017-07-29 @ 00:47:23 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:47:23 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:47:23 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:47:23 -- Board: OMNI, version: 0
2017-07-29 @ 00:47:24 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:48:04 -- EEPROM saved
2017-07-29 @ 00:48:04 -- Device - Rebooting
2017-07-29 @ 00:48:04 -- Serial port successfully closed
2017-07-29 @ 00:48:06 -- Serial port successfully opened with ID: 6
2017-07-29 @ 00:48:06 -- Serial port successfully closed
2017-07-29 @ 00:48:22 -- Serial port successfully opened with ID: 7
2017-07-29 @ 00:48:23 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:48:24 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:48:25 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:48:25 -- Board: OMNI, version: 0
2017-07-29 @ 00:48:26 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:50:05 -- EEPROM saved
2017-07-29 @ 00:50:06 -- Unrecoverable failure of serial connection, disconnecting...
2017-07-29 @ 00:50:06 -- Serial port successfully closed
2017-07-29 @ 00:50:07 -- Serial port successfully opened with ID: 8
2017-07-29 @ 00:50:08 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:50:09 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:50:09 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:50:09 -- Board: OMNI, version: 0
2017-07-29 @ 00:50:09 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:50:41 -- Serial port successfully closed
2017-07-29 @ 00:53:48 -- Serial port successfully opened with ID: 9
2017-07-29 @ 00:53:49 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:53:50 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:53:50 -- Running firmware released on: Apr 3 2017 22:14:20
2017-07-29 @ 00:53:51 -- Board: OMNI, version: 0
2017-07-29 @ 00:53:51 -- Unique device ID received - 0x34003a5633570a20373536
2017-07-29 @ 00:55:08 -- Unrecoverable failure of serial connection, disconnecting...
2017-07-29 @ 00:55:08 -- Serial port successfully closed
2017-07-29 @ 00:55:15 -- Serial port successfully opened with ID: 10
2017-07-29 @ 00:55:16 -- MultiWii API version received - 1.31.0
2017-07-29 @ 00:55:16 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-07-29 @ 00:55:17 -- Running firmware released on: Apr 3 2017 22:28:08
2017-07-29 @ 00:55:18 -- Board: SPEV, version: 0
2017-07-29 @ 00:55:18 -- Unique device ID received - 0x3d00275633570320313639
[... 511 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

derekmoc, 2017-07-29:
Thanks jflyper, that was a great help, at first I couldn't get it to work then I realized I still had smartport turn enabled on UART2.  Also changed 'resource serial_tx 11 a8' to 'resource serial_tx 11 A08' as that what was allocated to the led_strip, not sure if I needed to though.

Thanks again for your help, feeling the Betaflight love.

jflyper, 2017-07-29:
@derekmoc All set now?

derekmoc, 2017-07-29:
Works great now, thanks again for the help.  Have put a link to this on rcgroups

closed 2017-07-29 by derekmoc