betaflight/betaflight #3653: Spracing f3 evo brushed issue

Jdroid9, 2017-07-28:
I accidentally flashed my sp racing f3 evo brushed with the f3 evo binary.  I went and configured it with no issue before I realized that I flashed the wrong binary.  So I went back and flashed the f3 evo binary and now it is locked up and will not connect to my flight controller.  I have tried re-flashing it several more times.  Is it recoverable or is it dead?

jflyper, 2017-07-28:
@Jdroid9 Is there a firmware (binary) for "sp racing f3 evo brushed"? Never seen one before.

Jdroid9, 2017-07-28:
You are supposed to flash the f3 evo brushed with the f3 evo binary.  And I meant to say that it wont connect to the configurator now.  Sorry about that.

jflyper, 2017-07-28:
What are the nature of binaries that is wrong and correct?

And ... I'm still confused. Is this what happened?

(1) After the first flash with wrong one, you could connect to your FC.

(2) Then flashed with correct one, and you couldn't connect to your FC any more.

(3) After that, you can flash the correct one, presumably using BOOT0 button/pad, but you can't connect to your FC.

Jdroid9, 2017-07-29:
That is exactly what happened.

jflyper, 2017-08-05:
I suspect the "correct one" is not correct at all ...

Jdroid9, 2017-08-05:
You are correct.  I flashed it with the sp racing f3 code instead and it worked.

jflyper, 2017-08-05:
All set?

Jdroid9, 2017-08-06:
Yep.

jflyper, 2017-08-07:
Okay, happy props!

closed 2017-08-07 by jflyper