betaflight/betaflight #3654: BF3.2 DYS F4 not flying

pango, 2017-07-28:
I've installed the hex file for the omnibusf4 Target on my DYSF4 from Build #4 .
All RC Inputs are visible in the receiver tab. And Throttle in the OSD.
After the quad is armed it does nothing.
It does not react to any RC input.
Blackbox Log and CLI Dump can be found here:
https://drive.google.com/drive/folders/0B3_cYIppuywuemZ0TWl1Rll3NE0?usp=sharing

wind0r, 2017-07-28:
Configuration Issue.
Cutoff is set to 0. Arming isnt possible which that value,

pango, 2017-07-28:
Ok, looks like I messed up the cutoff and the normal filter value.
I chaned it and now the quad reacts to my input.

pango, 2017-07-28:
Maybee one should prevent users like from dooing that by not allowing a value of zero for the notch filter cutoff?

closed 2017-07-28 by pango