betaflight/betaflight #3657: Some OSD settings not available in CLI?

Arakon, 2017-07-29:
Horizon Sidebar setting doesn't seem to appear in a diff or full dump, or "get".
Same goes for the timer setting (warning time i.e. for armed time), it's always 10 by default even after pasting a dump.
I can change them by using the configurator.

DanNixon, 2017-07-29:
What firmware version are you using (if its 3.2 then what Git revision)?

Arakon, 2017-07-29:
Literally any Git revision. It's been happening in any build of 3.2.0 for several weeks for me, last one I tried it with being 402de03bd402ff6d6e5260dff404e934118cfeb3.

DanNixon, 2017-07-29:
Timer configuration is stored in the `osd_tim1` and `osd_tim2` settings, this stores the entire configuration (source, precision and alarm threshold) so is not easily manipulated via the value alone. (it is done this way to save flash space)

Horizon sidebars are missing, I'll add that now.

PR #3664 (2017-08-01): Add missing AHI sidebar setting to CLI
Fixes #3657

Adds the missing setting required to toggle the AHI sidebars.
closed 2017-08-01 by martinbudden