betaflight/betaflight #3669: Configurator 3.1.2 Rateprofile selection is broken

chime13, 2017-07-29:
Symptom:
Under PIDs tab, Rateprofile box defaults to last setting and remains there no matter what input is made.
Normally, I can toggle rates from my TX using an assigned AUX. 3.1.2 no longer responds to Rateprofile selection.
I did do not have this issue on 3.1.1 and confirmed in 3.1.2
Downgraded to 3.1.1 until issue is resolved

DieHertz, 2017-07-30:
Thanks for your report.
BF configurator has it's [own](https://github.com/betaflight/betaflight-configurator) repo, please report configurator-related issues there.

chime13, 2017-07-30:
I used the link provided in BFC to report the issue. (Last line in “Contributing” paragraph)  Maybe someone should write the correct link to avoid these issues