betaflight/betaflight #3672: build environement broken after STM driver update

adrianmiriuta, 2017-07-30:
---------------------------------------
%% (size optimised) vtx_rtc6705.c 
make[1]: *** No rule to make target 'lib/main/STM32F30x_StdPeriph_Driver/src/stm32f30x_comp.c', needed by 'obj/main/KISSFC/stm32f30x_comp.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
%% (size optimised) vtx_tramp.c 
----------------------------------------

AirBreak69, 2017-07-30:
I got rid this by simply deleting all intermediate files below the \obj\main folder.
By this you make a clean compilation.
Additionally you can do: make clean

DieHertz, 2017-07-30:
This is indeed caused by leftovers from your last build before updating. This is generally the case with building any software, whenever you update toolchains do a clean

closed 2017-07-30 by DieHertz