betaflight/betaflight #3673: Accelerometer reading on F4's in configurator/sensor: factor 2

AirBreak69, 2017-07-30:
Don't now, if correctly placed here and really relevant:
Noticed that in configurator/sensor the accelerometer readings are doubled on F4's (DYSF4PRO, REVO).
F3/F1's (LUX_RACE, NAZE) are OK.
My "fear": If not only a scaling problem towards MSP the levelling algorithms may be affected.
In blackbox files the values are OK (1g=1g).
Checked with the latest build 1523 but occured months ago.

jflyper, 2017-07-30:
I've confirmed it with F7, F4 and F3. Only F1 is unaffected.

Since the blackbox is showing the correct value, I suspect it caused by conversion process in MSP.

jflyper, 2017-07-30:
Wait... It's acc/gyro chip dependent...

jflyper, 2017-07-30:
It's this one
Added runtime setting of gyro SPI pin #2586

But I don't see anything peculiar...

AirBreak69, 2017-07-30:
@jflyper Thanks for investigating. You may be right, indicates to MPU6000+SPI
DYSF4PRO: ACC=2g, MPU6000, SPI? (no I2C listed in resource)
REVO: ACC=2g, MPU6000, SPI (board schematic), I2C for mag/baro
AIRBOTF4: ACC=2g, MPU6000, SPI? (no I2C listed in resource)
LUX_RACE: ACC=1g, MPU6500, SPI? (no I2C listed in resource)
OMNIBUS(F3,OSD,bruhed): ACC=2g, MPU6000, SPI? (no I2C listed in resource)
SPRACINGF3EVO (bruhed): ACC=1g, MPU6500, SPI? (no I2C listed in resource)
NAZE (Acro32): ACC=1g, MPU6050(chip marking, gyro type not shown!), I2C (in resource)

AirBreak69, 2017-08-01:
I checked with #3679 changes on DYSF4PRO, REVO, OMNIBUSF4SD.
@jflyper @martinbudden Thanks for digging deeply.

closed 2017-08-01 by AirBreak69