betaflight/betaflight #3674: OSD but no Video

agmorpheus, 2017-07-30:
I have the betaflight f3 (software version 3.17), a TBS Unify Pro HV and a Foxeer Cam. I wired all like on this plan: https://blog.seidel-philipp.de/wp-content/uploads/2017/03/Betaflight_F3_Flight_Controller_Anschlussplan_Wiringplan.jpg
The problem now is, that I have a clear OSD but only a snowy video. I don’t know whats the problem. Can someone help me?

wind0r, 2017-07-30:
Sounds like broken wire or cam.
Since the OSD is working the chip is fine.

jflyper, 2017-07-30:
@agmorpheus Betaflight github repo issue is for reporting and discussion of making Betaflight software. For issues related to particular hardware setup, please consider using appropriate thread in RCG.

https://www.rcgroups.com/forums/showthread.php?2795213-Betaflight-F3-Flight-Controller-OSD-PDB-SD-card-BEC-current-sensor

closed 2017-07-30 by jflyper