betaflight/betaflight #3676: BF 3.2 - props spinning full throttle for 3-4 seconds without arming

RipperDrone, 2017-07-30:
Any idea which state could produce this? 

Just found one BF early adopter in Germany reported deep prop cut injuries, he's in hospital now... I asked him for his diff and setup to know more...

Have BF 3.2 running on 3-4 quads now, no such thing ever happened...

Ouch... :-(
[image]

wind0r, 2017-07-30:
Those pictures arent appropriate in a bug tracker.

Without Information nothing can be done.
Some Radio misbehaving or using the broken binary from yesterday which didnt saved configs.

RipperDrone, 2017-07-30:
Pic deleted. Some information has been shared in the meantime:

- he did a full flight, no probs
- landed, disarmed on switch -> motors off
- RC was still on. He approached the quad (presumably with the RC in his hand) and lifted it up from the ground
- all of a sudden props started spinning high for a few secs, then stopped

Sounds like Failsafe might have kicked in due to proximity of RC to quad (bad signal) even though quad was disarmed. Is there any scenario that Failsafe might engage in disarmed state?

Arakon, 2017-07-30:
I've been flying 3.2.0 on 3 different quads in the past few weeks and never saw anything like this happening.
That includes picking up the copter in the same hand as I was carrying my transmitter (turned on), so it was literally inches from the receiver.

DanNixon, 2017-07-30:
@RipperDrone I was about to say "failsafe can never arm a disarmed quad", however after testing it appears that if a user sets their stage 1 failsafe to set the aux channel used for arming to within the ARM box range then the quad will arm if the RX link is lost.

This may have been what happened, however I'd expect this to come up in bench testing.

I'd like to see his `diff` if you get it to confirm this.

RipperDrone, 2017-07-30:
nuffn special...

# diff
# Betaflight / SPRACINGF3 3.2.0 Jul 20 2017 / 15:41:34 (20748dc)

feature -RX_PPM
feature -TELEMETRY
feature -TRANSPONDER
feature RX_SERIAL
map TAER1234
serial 2 64 115200 57600 0 115200
aux 0 0 2 900 1300
aux 1 4 2 900 1300
aux 2 13 0 900 1300
aux 3 26 1 1650 2100
aux 4 28 2 900 1300
set gyro_sync_denom = 2
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 36
set yaw_motors_reversed = ON
set pid_process_denom = 1
profile 1

set i_pitch = 35
set i_roll = 25
set i_yaw = 20
rateprofile 0

DanNixon, 2017-07-30:
Unless it was just bad failsafe settings on the RX/radio then. I don't use Spektrum so not sure what it would default to.

DieHertz, 2017-07-30:
There actually is something special, this guy (and probably other Spektrum users) has ARM on 900-1300.
When his Rx goes into failsafe, it may as well output a 900-something signal on aux channel instead of no-pulses or low-width pulses.

swilkens, 2017-07-30:
AFAIK Spektrum receivers program their failsafe positions for all channels during the binding:
https://www.spektrumrc.com/Technology/SmartSafe.aspx

If we assume he had his arm switch on low during the bind and further assuming he had the ARM mode in the firmware set to the low range, a loss of signal on the receiver could drop the channel to the failsafe and trigger the ARM.

RipperDrone, 2017-07-31:
...but it still sounds like all the AUX channels should hold their last status (which was disarmed state as per his description)

swilkens, 2017-07-31:
After hooking up my Spektrum sat receiver I can confirm that a loss of signal (turning off the radio) will result in:

Throttle - Fully low regardless of failsafe or previous stick position.
Yaw, Pitch, Roll - Positions set at binding stage (mid stick in my case)
AUX1 / AUX2 switches - holds the last command, doesn't seem affected by receiver fail-safe

This doesn't match with the website which states that all channels other than throttle will hold their last command. In my testing, all **stick** channels other than throttle will jump to their fail safe position. AUX channels (switches) appear to hold their last command.

Assuming the guy was using a switch for the arming sequence, I don't see how a loss of signal could trigger an arm unless he somehow accidentally moved a switch... or if fail safe behavior has changed over time. The only way to be sure would be to check his receiver.

Back to the topic at hand, I wasn't able to trick the quad into arming after entering stage 2 failsafe. If stage 2 was disabled, however...

wind0r, 2017-07-31:
@swilkens your result is exaclty what betaflight is doing when a failsafe is detected but i think we also need to check the Spektrum Radio Config.
I think similar to Frsky you can enable a failsafe mode on raido/receiver where channels got changed to user defined values. That could arm a quad.
Every Receiver should be bound with Failsafemode "No Pulse" (i dont know the spektrum name for that)

swilkens, 2017-07-31:
@wind0r Clearly I haven't had my coffee yet, previous results was stage1 setting the channels - not the actual receiver. 

Doh! I'm hooking up a scope to a separate receiver to understand the actual fail safe behavior.

Additionally - I wouldn't be surprised if a satellite receiver doesn't offer fail-safe features, as these are usually connected to a main receiver. Do we know the exact receiver the guy was using? I'll verify with an original satellite as well...

gubernator46, 2017-07-31:
Don't know if this information will be helpful:
We had a similar issue with the quad of my Kids lately. The quad was disarmed and one of the twins had the receiver in his hand, while the other one was picking up the quad.
For about a second (maybe shorter) the motors startet to spin.
Fortunatly the one holding the quad droped it fast.
After that I ranted the one holding the receiver, because I thought he accidently touched the arm switch, but he said he didn't touch it.

It was on Betaflight 3.1.7 with a Spektrum system and a 900-1330 arm range.
Here is the whole diff:

# diff
# Betaflight / SPRACINGF3MINI 3.1.7 Apr  3 2017 / 22:28:52 (e1c4b5c)

feature -RX_PPM
feature -BLACKBOX
feature -SDCARD
feature RX_SERIAL
feature TRANSPONDER
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
aux 0 0 1 900 1300
aux 1 28 0 1300 2100
aux 2 4 0 1300 2100
aux 3 13 0 1700 2100
set max_throttle = 1955
set min_command = 950
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = SBUS
set vbat_scale = 108
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 34
set align_board_yaw = 180
set gyro_sync_denom = 2
set acc_hardware = NONE
set pid_process_denom = 1
profile 0

set p_pitch = 55
set i_pitch = 40
set d_pitch = 25
set p_roll = 48
set i_roll = 30
set d_roll = 20
set p_yaw = 85
rateprofile 0
rateprofile 0

set rc_rate_yaw = 150
set rc_expo = 30
set rc_yaw_expo = 30

jumatic, 2017-07-31:
Had this issue also, but only one single time it happend, some time ago. And i am also a Spektrum user. Dont know which firmware it was, always when connecting up the Naze to BF with battery, the motors started spinning for 1 or 2 seconds while disarmed with multishot. Then a new FW came up and this problem was gone. 
After that i finished my built on a 400 quad with 8" props, made a testflight, disarmed, motors stopped. When i was trying to pick it up, motors started spinning. throwed it upside down to the ground. It was in my case with using 3D mode, think i was trying setting up failsafe working right in 3D mode. 
Never had this issue again with stock failsafe settings and never used 3D mode again ;-)
Maybe i can reproduce that, will try on a small quad.

jumatic, 2017-07-31:
Tried some scenarios. Only thing what i found is, if you adjust the failsafe valid pulse range near to the aux switch disarm range, and also have minimum throttle at 1000 where the motors dont spin, you pull down throttle stick and disarm, it is not. So you may think its disarmed and try to pick it up it will spin up. It will spin faster certainly with airmode active. I think this happend to me. So my advice, set the endpoints right, at my spektrum the aux switches endpoints are set to 148 and have a pulse range at 995 at low position. Stock valid pulse for failsafe is set to 885 so it wouldnt happen. 885 to 900 is very close. At my test i have set failsafe pulse range to 992 and disarm havent worked. Maybe some people doesnt understand the valid pulse range at the failsafe settings, like me when i was starting with this hobby. 
Maybe its just a endpoint setting failure????

RipperDrone, 2017-08-01:
Closing, since PRs have been created / merged by DanNixon to double-lock arming and to introduce toggle arming need when throttle is >min (to avoid unintentional arm when throttle is lowered back to min)

closed 2017-08-01 by RipperDrone