betaflight/betaflight #3677: F4 boards and spektrum receivers

dennismatovich, 2017-07-30:
Currently experiencing trouble using f4 revolt v2, f4 dys, f4 matek boards with spectrum receivers. The 2 receivers telemetry and non-telemetry both work perfect on f3 boards, these receivers will only light up solid when attached to the f4 board without a transmitter being on.  All tech support suggestions fail with same result or no light on receiver at all. Not once has the receiver entered "bind mode" flashing amber light.  

- All three boards were not installed on quad, only receiver was attached and usb 
- Tried 2 different computers with bf 3.1.2
- Flashed with suggested firware
- ppm works fine as tested with old frsky receiver and transmitter

jflyper, 2017-07-30:
@dennismatovich If you are trying to get your receiver into bind mode while connected to F4's serial RX connector, which usually have uni-directional inverter, it won't work.

This is a limitation brought by F4 design and board design. BF3.2 will give you some option to overcome this limitation (but not universal, since uni-directional inverter can't be modified by software to do bi-directional inverting).

dennismatovich, 2017-07-30:
Thank you @jflyper for response.   I can bind the receiver and radio without the f4 boards, when wired to these boards the link is broken between receiver and radio, the problem exists without binding...

In my quest to finish this build with an f4 board and spectrum receiver combo, my findings are: The Omnibus F4 Pro team figured this problem out and have a specific spectrum jst port, when the receivers are connected to this port they act normal. Seems like the sharing of ports for multiple receiver types is the problem, this maybe what you were saying in tech talk.

jflyper, 2017-07-31:
@dennismatovich You mean you just can't receive any RC input with spek?

dennismatovich, 2017-07-31:
on three different f4 boards there is no communication between receiver and transmitter, on the omnibus f4 pro it works great.  The 2 receivers do not have srxl and they are not satellite receivers, if that helps...

jflyper, 2017-07-31:
@dennismatovich 
> there is no communication between receiver and transmitter

Are you saying
(1) RC receiver does not link with RC transmitter, or
(2) RC receiver links with RC transmitter, but there is no RC command stream received by FC?

dennismatovich, 2017-07-31:
1

jflyper, 2017-07-31:
@dennismatovich Doesn't that mean your receivers are not bound to transmitters? I'm not a Spek user, but my guess is that receivers links to transmitters regardless of FC presence if they are bound properly.

dennismatovich, 2017-07-31:
I believe the autobind feature with the two receivers changes how they work with FC. Normally a bind plug is used and no problems with most spectrum receivers. When autobind receiver is used now the fc controls the state of bind not the bind plug, which is great until f4 boards popped up...

jflyper, 2017-07-31:
@dennismatovich 

Can we talk on a single FC and simple satellite receiver to make this conversation more efficient?
Which setup uses satellite? Or, do you have any preference on which FC you want to tackle first?

dennismatovich, 2017-07-31:
I'm looking forward to the new release. 

Yes great idea, I have the matek f405 on my bench and a spm4648 spectrum serial quad receiver. 

Side note: I received support from matek on spectrum receiver issue,  using a remote satellite receiver in dsm2 mode was successful on rx1 rx2 rx3 rx4 and rx5.

jflyper, 2017-07-31:
> Side note: I received support from matek on spectrum receiver issue, using a remote satellite receiver in dsm2 mode was successful on rx1 rx2 rx3 rx4 and rx5.

Okay, so far, so good. What's next? 4648 in telemetry mode?

dennismatovich, 2017-07-31:
the 4648 is basic no telemetry, its a 3.3v serial receiver, reason for use is lightweight and has antenna diversity.  I will wire it to 3.3v, ground and rx2 for first test.

dennismatovich, 2017-07-31:
Went to chrome store launched betaflight configurator 3.1.3, connected to matek f405 and setup normal spektrum settings: **ports page** - port uart2 as serial rx, **config page** - spektrum serialrx, spektrum 2048, blackbox, osd.
save and reboot. The picture attached is result - no light at all on rx.
  The dx8g2 transmitter is not turned on during tests.

dennismatovich, 2017-07-31:
[image]
Looks like after wiring this so many times I switched the 3.3 and signal wire. Found this a minute ago to check my connections...  So after that I now can bind to the receiver but in dsm2 only, that is a success to some level but not dsmx yet.  firmware - 
 version
# Betaflight / MATEKF405 3.1.7 May  2 2017 / 19:59:41 ()

jflyper, 2017-08-01:
Go for 3.2RC1, and after flashing, enter this cli command

```
resource rx_bind a3
```

You may want to try this also
```
resource rx_bind none
```

Ref:  
https://github.com/betaflight/betaflight/wiki/Spektrum-Satellite-Bind-for-3.2

dennismatovich, 2017-08-01:
Before entering those I checked my radio and manually selected dsmx 11ms for this model and tried binding with success on dsmx 11ms.  This f405 is now communicating properly with spm4648 spectrum receiver using betaflight configurator 3.1.3 and matek hex firmware.

jflyper, 2017-08-01:
Cool ;)

closed 2017-08-02 by dennismatovich