betaflight/betaflight #3680: What could be causing the motors to move randomly when the dron is disarmed?

partoco, 2017-07-30:
https://www.youtube.com/watch?v=uoZ-hZpFmBA

Quad: Armattan chamaleon
Fc: cl_racing f4
Receiver : turnigy ia6c
motors:Armattan OOmph 2206/2300 KV
betaflight 3.2.0

RipperDrone, 2017-07-30:
Which motor protocol?

Typically had this when ESCs were not calibrated correctly and even "zero" motor output was above ESC min_PWM so motors started twitching...

Ozzy33, 2017-07-30:
You left out an important piece of info, what esc and esc FW are you running? If they are Atmel based, this is the old Atmel chirp bug, lower min_thro to ~980 or even lower.

DanNixon, 2017-07-30:
What is the Git revision of the firmware you are using, there used to be an issue with jitter on the motor outputs affecting analog motor protocols earlier on in 3.2.

partoco, 2017-07-30:
Hi all, the protocol i am using is dhot 600 and the firmware of the esc is the ch25, 16.6, but i think i found the bug.

The first thing that I have done is to make sure that I have the channels well separated from each other and do not overlap between them. After I have separated the buzzer (it comes integrated in the racing of armattan) Of the arming channel, leaving it only one chanel to arm and disarm. And the buzzer I have taken to the channel where I had the horizon mode. Now when disarming the quad, the engines no longer do crazy, but as soon as they activate the buzzer, even being disarmed, does it again.
To that I believe that the fault is related to 100% with the integrated buzzer. I'm a rookie in this, to if I can be wrong Regards, and thank you very much.

DanNixon, 2017-07-30:
@partoco So it only happens when the buzzer is on?

In that case it may just be a result of the beeper being forwarded to the ESCs (via DShot commands).

Usually a small amount of motor actuation is expected but not enough to actually spin the motor, you could try reducing the strength in BLHeli (IIRC it is called "beacon strength" but check this before making any changes).

partoco, 2017-07-30:
Effectively, it occurs only when the buzzer is active. Thank you, I'll look at what value it is

partoco, 2017-07-30:
The value of beacon strength is 80

Ozzy33, 2017-07-30:
go update your BFC, they just updated it this morning.

partoco, 2017-07-30:
Yes thanks. Was updated to 3.1.3, but the same thing is still happening

RipperDrone, 2017-07-31:
Have you tried setting min_throttle to <980?

RipperDrone, 2017-07-31:
verified your RC in Betaflight Configurator that it actually outputs a proper range from 1000-2000us?

RipperDrone, 2017-07-31:
set min_check to >1050?

partoco, 2017-07-31:
Hello ripper. betaflight configuration, this is how I came from armattan. I'm going to check that

DanNixon, 2017-07-31:
I think we should rule out the DShot beeper before suggesting potentially irrelevant fixes.

@partoco Try setting the beeper strength in BLHeli to zero and testing again, this should disable ESC beeping using the motors and will confirm if my theory was correct.

alfag4, 2017-07-31:
Some days ago when i installed BF 3.2 i had the same issue.
My FC is an Omnibus F4 Pro V3 - DShot600 with external buzzer. After some investigations i discovered that removing the buzzer from the Aux channel solved the issue but in this manner i cannot use the beeper so i revert to 3.1.7.

Try to completely remove the beeper from any aux channel

partoco, 2017-07-31:
DanNixon, I'm going to prove that having happened, thanks for your time, really

alfag4, that is the temporary solution that I have chosen. Separate the buzzer from the arming channel and use it only when necessary. a greeting

richard-scott, 2018-02-07:
I get movement on my motors with Blheli_S Dshot600 and ESC beeping. The motors jitter in time with the beeps. Isn't that just normal? If I turn the beep volume down in the Blheli Suite the motors don't move. If I turn up the beep volume they jitter more.

I have always assumed that as the motors do not have an internal beeper that the noise is created from motor movement.

richard-scott, 2018-02-07:
I don't have the buzzer on my arming channel BTW.

mikeller, 2018-02-08:
@richard-scott:

> I get movement on my motors with Blheli_S Dshot600 and ESC beeping.

That's quite normal, since the motor's magnets are used to vibrate the bell when beeping. Some motors will move when this is done.

richard-scott, 2018-02-08:
@mikeller I figured that was all normal :-) I just wanted to add in my comments so hopefully the OP could see that this is normal for motors to move with dshot beeping.

closed 2018-04-21 by DieHertz