betaflight/betaflight #3684: BF 3.2 motolab has no motor #1 signal

DERBYJ, 2017-07-30:
No motor one signal one motor #1 pad for Motolab target

# resource list
IO
A00: BEEPER
A01: FREE
A02: FREE
A03: FREE
A04: MOTOR 1
A05: ADC_BATT
A06: MOTOR 2
A07: FREE
A08: FREE
A09: I2C_SCL 2
A10: I2C_SDA 2
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: MPU_EXTI
B00: MOTOR 3
B01: MOTOR 4
B02: FREE
B03: FREE
B04: FREE
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: FREE
C14: FREE
C15: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: MOTOR 3
DMA1 Channel 3: MOTOR 4
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 2
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

#

jflyper, 2017-07-31:
@DERBYJ 

TIM3 CH2, assigned to PA4 (motor 1), does not have DMA capability, and that is why motor 1 does not work with Dshot if this is your case, and the board wiki has a line for this.
https://github.com/betaflight/betaflight/wiki/Board---MOTOLAB#faq--known-issues

If this is not your case (non-dshot), please do a full chip erase when you flash, or execute "defaults" on cli and see if it makes any difference.

DERBYJ, 2017-07-31:
@jflyper 
I will just assign pin like I did on first FC I flashed. 
Thank you much!!!!

closed 2017-07-31 by DERBYJ