betaflight/betaflight #3690: BF 3.2.0 - custom Firmware without ESC-Sensor

TheIsotopes, 2017-07-31:
i'm testing a firmware without ESC Sensor ... removed ` #define USE_ESC_SENSOR` from `common_fc_pre.h` and `target.h` ... but if i going to bf configurator power & battery tab and using `Onboard ADC` i become a very long `Voltage Meter` list (see sreenshot) ... maybe there are over 200 entries of `Battery` and the `Serial link status` indicator flashes like crazy ... is this a bug from firmware or configurator

[image]

basdelfos, 2017-07-31:
Most of the credit about the new Power & Battery tab goes to @hydra ;)
But I recently did a rework of layout for the tab. 

I will have a look into this to verify it is a configurator or firmware bug. 

I expect that the serial link status flashing like crazy is a result of the long list of battery entries.

basdelfos, 2017-07-31:
Something definitely goes wrong in the fw. Current firmware master reports 249 voltage meters with `MSP_VOLTAGE_METERS` when `#define USE_ESC_SENSOR` is removed or undefined.

basdelfos, 2017-07-31:
Btw: why do you need a firmware without ESC Sensor?

TheIsotopes, 2017-07-31:
> Btw: why do you need a firmware without ESC Sensor?

Only to test a super stripped small firmware ... everything deactivated which i not use ;-)

> Something definitely goes wrong in the fw. Current firmware master reports 249 voltage meters with MSP_VOLTAGE_METERS when #define USE_ESC_SENSOR is removed or undefined.

when testing the firmware i noticed this issue ... can you fix this or we need to ask someone else?

basdelfos, 2017-07-31:
I can fix this, it's on my todo list

TheIsotopes, 2017-08-05:
@basdelfos any news about fix this issue

basdelfos, 2017-08-30:
Latest news (before my holiday) was that I managed to reproduce the issue. Only did not have time to look further into a solution. It is a low priority issue as we first want to release a stable BF 3.2.0 with configurator and this issue is really specific to your case.

TheIsotopes, 2018-02-05:
@basdelfos are already successes here or it's fallen under the table :((

basdelfos, 2018-02-08:
@TheIsotopes yes this totally has fallen under the table. But I remember there where some changes made by the devteam round this subject. You could try first if this has been fixed yet?

TheIsotopes, 2018-02-10:
nope ... nothing has been fixed yet today

TheIsotopes, 2018-03-27:
fixed now in #5547

closed 2018-03-27 by TheIsotopes