betaflight/betaflight #3692: Tramp HV incompatibility issue with F4 hardware UARTs

jflyper, 2017-07-31:
Issue reminder.

There has been multiple recent reports of Tramp HV not working with F4 hardware UARTs. Most claims that NONE of the UARTs work. The VTX itself works without a problem when connected to a software serial port.

Saxin, 2017-08-02:
Same here - I confirm that Tramp HV doesn't work on HW UARTS on Omnibus F4SD target , works on soft serial, TBS unify pro doesn't suffer same issue.

Related to #3329 .

the-scriptinator, 2017-08-04:
I have a maybe similar problem with the Matek AIO board.

Tramp telemetry was on TX1 and smart audio did not work.
Tested TX1 solder joint to VTX plug on the other end of the cable for continuity on the telemetry line and continuity was OK.
Moved the telemetry line to TX5, config'd in ports tab and Tramp Smart Audio  worked fine.

(OK it's not really smart audio since it is not a TBS product, but what is the Tramp telemetry feature called?)

bonhardt, 2017-08-04:
i have omnibus f4sd v2 and f4 normal, both works fine on hardware uart, 3.1.0 3.1.6 and latest 3.2.0

ep092, 2017-08-05:
Tramp HV ist working with Fortini F4 and v3.2 at startup but when Tramp reaches 80 degreee Celecius the FC freezes and stops the motors. If I disable Tramp in the Ports menu everything ist fine even at hotter temperatures.

Gow864, 2017-08-22:
I'm using a Matek F405 board with  Matek HV Vtx. Using uart 4. In this setup the Tramp telem does not function....

wallbanger123, 2017-09-10:
Same problem here on the Omnibus F4 Pro corners. Tried UART1 and 6 ( subs is on 3 with this board)

jflyper, 2017-09-10:
@wallbanger123 UART6 on Omnibus F4 Pro Corners has uni-directional inverters, so it does not work.

wallbanger123, 2017-09-10:
It doesn't work on UART1 either.

Saxin, 2017-09-10:
@wallbanger123 - please educate yourself with this wiring proposal by @jflyper (Teralift on RCG).
[https://www.rcgroups.com/forums/showthread.php?2949247-OMNIBUS-F4-Pro-Corners-Configuration-Example](url)
Note that TX1 pad is used via soft serial.

This is a standing issue on F4 boards, very easy to work around - using soft serial on some unused pins.

wallbanger123, 2017-09-10:
I'll give that a try. Thank you

Poitivin, 2017-10-23:
Is this open bug still being investigated for a resolution?

I moved pretty recently from a CF naze32 to an SPRacingf4EVO with BG 3.2.1 and came across this issue with my TrampHV. From the reading I understand I could potentially use an workaround by setting a softserial, but it happens (not sure why, although posted today in the RCGroups BF the question) my SPRacingF4EVO does not enable softserial and with that said in alternative I was trying to explore the resources re-mapping, although not sure if it works and how I would end up telling the BF that the peripheral TrampHV is connected there, like remap motor 5 to NONE and set the resources serial_tx 6 or 7 or any higher number to that old motor PIN....

jflyper, 2017-10-23:
@Poitivin You have been answered extensively in the thread.

Poitivin, 2017-10-24:
Thanks, The SPRacingF4EVO has 1 VCP and 5 UARTs, and as much I try to use the resource to set SERIAL_TX 6 XX, where XX was the resource allocated to other PIN, like motor 6 (which is now set to NONE), softserial feature either in the GUI or CLI does not get enabled. tried with other PINs, including LED Stripe and same, no softserial, so I guess softserial depends on the board and not trough the BF firmware. I´ll follow in the thread

Is this bug still to be sorted by getting the UARTs compatible with F4 boards?

Could it be the open bug https://github.com/betaflight/betaflight/issues/2823 for the softserials SPRacingF3 is somehow present in the SPRacingF4 board, although I may think the F4 board should not suffer from the same flash space?

Thanks
Fernando

jflyper, 2017-10-24:
Like everyone said in the RCG thread, resource index for softserial are 11 and 12.

I consider the incompatibilities as hardware problem, as there are contradicting reports of working/non-working cases on the same port on the same FC.

To investigate, someone with appropriate knowledge and resource must have a setup that doesn't work, and this is why the issue hasn't been worked on up until today.

Poitivin, 2017-10-24:
Thank you. I´ve missed the 11 and 12 on the serial_tx because I didn´t knew and my CLI only show up to 10 and softserial never gets enabled, so cannot allocate the resources there.
I am getting the impression the f4 board suffers from the same as the f3 on the open bug I found, whereas it should not as being an f4 it should support the flash space (me thinking)...
I´ll follow with the BF thread, thanks for your time

Saxin, 2017-10-24:
@jflyper 

I have x2 setups with this incompatibility - let me know what you need to further investigate.
Alternatively email me: saxin1 (at) gmail (dot) com , I can send you one Tramp if you want to do it yourself ?

Poitivin, 2017-10-24:
Just for reference I got an answer from the SPRacingf4EVO target support (https://github.com/betaflight/betaflight/pull/2796) that softserial is not enabled in this board which unfortunately brings me to a limitation against the possible workaround for the TrampHV usage. Why it´s not enabled I don´t know although asked the question

Poitivin, 2017-10-30:
I´ve made further testing with an SPRacingF4EVO and TrampHV works has expected only at UART4, why only at this UART (excluding 2 and 5 due to the built in inverter and telemetry) I don´t know.

AmitShalev123, 2017-11-05:
tramp hv telemetry not working with the new betaflight f4 fc,  
when URT1 is configured as a regular URT nor as softSerial (as recommended by fpvmodel).
Tested with working vtx form another quad.
OSD is visible but unable to change/save settings in the vtx.

jflyper, 2017-11-06:
@AmitShalev123 Let us see your CLI command "diff" output, as it may be a configuration problem.

Enclose the text with three back quotes on the line by themselves if possible.

AmitShalev123, 2017-11-06:
@jflyper 
Hi please see my "diff" output
```
# diff

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.2.0 Sep 28 2017 / 16:17:10 (ceb6f70) MSP API: 1.36

# name

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature SOFTSERIAL

# beeper
beeper -ON_USB

# map

# serial
serial 30 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2100
aux 1 1 2 900 1225
aux 2 13 1 1875 2100
aux 3 27 1 1850 2100
aux 4 28 2 1700 2100
aux 5 35 3 1800 2100
aux 6 36 4 1800 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT1200
set failsafe_kill_switch = ON
set current_meter = ADC
set battery_meter = ADC
set pid_process_denom = 1
set osd_vbat_pos = 44
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2486
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 2466
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF

# profile
profile 0

# rateprofile
rateprofile 0

# 
```

DaveKraft, 2017-11-08:
I have very similar issue with my Omnibus F4 Pro V2 and Matek VTX (with should use the same "protocol/communication" as Tramp HV). I have XSR, Runcam Split and frsky telemetry on HW UART ports 1,3,6, so I have to enable softserial port for this VTX, but it doesn't work in any combination! I try to remap PWM5 and PWM6 to serial_tx 11, and serial_rx 11, I tryed to swap them, use just TX (that should be enough for proper function) and nothing works. i eaven try to use softserial for frsky telemetry, and that didn't work either. PS: I use all latest versions (BF 3.2.1, latest XSR FW, etc.)

Saxin, 2017-11-08:
@Davekraft matex vtx hv has issue with soft serial, update it's FW via matek instructions & you should be good to go.

DaveKraft, 2017-11-08:
Saxin, thx for advice. Matek is brand new, I bought it yesterday so I assumed that it is on latest firmware, but I check it immediately when I get out of work. :-) thx again.

DaveKraft, 2017-11-08:
@Saxin, thx again! :-) you were right, I had old FW in my VTX, so I flash the new one and it is working like a charm! :-)

mavericm1, 2017-11-12:
i've been trying to figure out why tramp protocol isn't working between kakute f4 and my 5804 v2 vtx for the past 2 days.  soft serial reassignment hasn't worked it won't work on uart 6 tx i've tried soft serial on both the serial 6 tx pin and also motor 6 pin.  None of that has worked i started to feel like the vtx was bad so i took a known working vtx i use in another build put it in this one and it doesn't work on soft serial or the hardware uarts.  At this point it is looking like some sort of software bug to me the other build was a f3 board with betaflight 3.2.1 and working beautifully..  I'm willing to test and help you with whatever i can do.

svfalcon7, 2017-11-14:
We're getting some reports of people that are using betaflightF4. They are saying Tramp protocol on UART1 soft serial or no soft serial/ being used as normal uart will not work. But if they are using UART3 Tx that it is working on regular uart with soft serial NOT enabled they are able to get vtx protocol to work.

AmitShalev123, 2017-11-14:
Thanks *JasonFPVMODEL*
I will try it and report back

RealNightEagle, 2017-11-15:
I have unify working on original SP F4 EVO UART 3. On my clone F4 EVO I can not get Unify 5V to work. It did work once on original SP F4 EVO but now I use Unify HV on that and it works fine.
What is the conclusion with Unify 5V ? is there a fix? I tried the CF and BF combinations but that does not work. I will try the soft serial approach on my bench FC and see if that works tomorrow and post back with results.
I can say this, with all the testing and messing I have had Unify 5V working on both F4 EVO clone and original F4 EVO UART's but 5V version fails after re flash.

si618, 2017-11-21:
@svfalcon7 Hi Jason, I just ran into this and swapping UART1 to UART3 fixed issue with Tramp telemetry.

`Betaflight / BETAFLIGHTF4 (BFF4) 3.2.2 Nov 11 2017 / 00:21:54 (77782e602) MSP API: 1.36`

Running Crossfire Micro RX v2.11 on UART1 with full telemetry. Can post full dump|diff if it helps.

mavericm1, 2017-12-09:
i measured the resistance of the serial_tx uart 6 pin between the 5v rail and it had a 10M ohm value signifying that there was a pullup resistor already in place.  However testing the rest of the uarts showed a much lower value of 5.7k Ohm.

DieHertz, 2017-12-09:
5.7K looks much more like pull-up than 10M to me.
Just try it

mavericm1, 2017-12-09:
yeah 1k ohm to 3.3v pullup did work and tramp is working now man what a pita

jflyper, 2017-12-09:
Hm... It still is a pragmatic solution. We still have selectivity among different UARTs on the same FC, right?

mavericm1, 2017-12-09:
@jflyper it really depends i guess all other protocols have seemed to be working fine for me its mostly tramp that has seemed to be the issue.  on the kakute f4 that came with a furibee x215 i couldn't get the vtx to work on any uart even softserial however on another working quad i had that same vtx worked fine with tramp.  I then switched to using a akk vtx that is doing smart audio and it worked without issue.  This last issue was with a racerstar f4s tatto which uses the luxf4osd target on 3.2.2 and it had the same issue with a vtx that does tramp this time i tried soft serial on the same uart but i didn't move it to a different test another uart as i had all other uarts in use and it was suggested to use a 1k to 3.3v pullup resistor which fixed it.   So i really can't make heads or tails for whats going on .  it did seem interesting that the resistance measured for the uart it was on was different than other uarts when tested with a multimeter against the 5v rail.

jflyper, 2018-04-12:
Those of you having trouble with Tramp on hardware UART1_TX of majority of F4 boards (those with the TX1 mapped onto PA9) may find a working solution now; try latest nightlies.

Lazy8onez, 2018-04-29:
Hey,

Just wanted to share my experience with (I think) this issue.

I have a HGKRC F428 board it has tx3, rx3, tx6, rx6, sbus, ppm, led, dsmx pads.
Currently I am using:
tx3 for F.Port - working great.
tx6 for IRCTtamp - working great.

I would like to add a Runcam camera control board, which requires both tx and rx of a UART.

I decided to remap one of the free pads (rx3 and PPM) to SERIAL_TX 11 (soft serial) or SERIAL_TX 1, and move IRCTramp there in order to free UART6.

welllll.....  none of this works.

romecode, 2018-05-21:
Hi,

I can confirm that my Omnibus F4 Pro Corners does not talk to Tramp HV correctly. Like some instances online, it seems that PIT mode setting does respond.

I also tried moving from Serial_TX1 to LED, and using both SOFSERIAL 1 and 2 respectively, no luck, same behavior.

I have a scope so can probe if needed.

Thanks!

romecode, 2018-05-25:
Confirming that under OMNIBUSF4SD using M5 (A01) on SOFTSERIAL SERIAL_TX 11 made Tramp work for me.

ghostface, 2018-06-24:
Can confirm that with a Zeus F4 - softserial via PPM or RX6 both lead to only pit mode working but nothing else.

drprops, 2018-07-28:
Bump

I also can't change settings for the Tramp HV using the betaflight OSD. I have a Matek F405-STD board, and the telemetry wire from the tramp is soldered to the Tx5 pin on my Matek board.

Oddly enough, on BF 3.2.1, the Tramp HV protocol works, and I am able to change settings via the OSD. However, if I upgrade to BF 3.4, it no longer works.

andras, 2018-08-10:
Can't get Tramp to work either.  I tried various pull-ups, no success at all.  VTX can be placed in pit mode by the FC, but nothing else works, not even setting it back to normal mode (which now requires a power cycle).

Hardware: FrSky XSRF4PO, VTX is a Holybro Atlatl HV (V1 with Tramp) connected to the LED strip port (B06) using SOFTSERIAL 1.

Analysis with a scope reveals that a series of pulses is sent every second (swing between almost 0V and 3V), the VTX sends a reply immediately afterwards (swing between almost 0V and almost 5V).  I haven't been able to capture any change in the pulse sequence with various settings yet, though, it looks exactly the same whatever I try.

DieHertz, 2018-08-10:
This issue was about a specific problem with F4 UART1 which was fixed.
Your's is likely a configuration issue

dschiedsch, 2018-11-27:
Hi
I had the same problem
It seems that on F4 board compared to F3 boards when using the Open-Collector serial setting pull up is not automatically activated.
Some vtxs do not have an internal pullup and so its not working
I fixed this with following code changes:

in serial_uart_stm32f4xx.c line 399 i changed IOCFG_AF_OD to IOCFG_AF_OD_UP
then I had to define this setting in io.h
I added a new line at 63
#define IOCFG_AF_OD_UP       IO_CONFIG(GPIO_Mode_AF,  0, GPIO_OType_OD, GPIO_PuPd_UP)

Now the behaviour is the same as on F3 boards

BarendDS, 2019-02-26:
Hi Guys,

Any news on this issue? I just received a tramp vtx which I wired like I did wire my Matek VTX and it didn't work. I'm using a CLRACINGF4S so it's supposed to be on TX4. Like I said this worked for my Matek VTX but doens't work for the tramp. I tried flashing to the latest Betaflight release and replacing TX4 with TX11 (softserial) but that also doesn't work. 

Tx in advance.

haslinghuis, 2022-05-10:
Closing as inactive. Please reopen if new data arrives

closed 2022-05-10 by haslinghuis