betaflight/betaflight #3695: Kakute F4 SmartPort getting worse over time

digaus, 2017-07-31:
Have the Kakute F4 and afte while I do only get sporadic values... same Bug seems to be present on the AOI version (same target). Have latest reciever FW and reciever was working fine another F4 before. So this seems to be a problem with the build in inverter of the Kakute board. Either software or hardware (I hope SW). Another user made this video showing the problem: https://youtu.be/yOPyDSde-eA

I am running the latest build...

jflyper, 2017-08-01:
@digaus Have this setup worked before? Is the receiver is full working condition? I'm trying to isolate the source of the problem.

digaus, 2017-08-01:
Not with the Kakute F4 nur with an Omnibus F4 and an F7. On Kakute it stops working after some time.

digaus, 2017-08-01:
Also when Softserial is active (Motor 5 and 6) the LEDs are behaving odd.

jflyper, 2017-08-01:
@digaus Your motor 5 output have timer collision with LED_STRIP. You can't use motor 5 nor softserial over this port when LED_STRIP is enabled. If you are running SmartPort over software serial and the software serial is on motor 6 (provided you are not using PPM; PPM also has a timer collision with motor 6).

```
resource serial_tx 11 c8
save
```

digaus, 2017-08-01:
I am running Softserial for the Split. Thats why I have RX in TX SS or is the split able to communicate bidirectional?

digaus, 2017-08-01:
Could I use motor 5 for RX and motor 6 for TX?

jflyper, 2017-08-01:
> I am running Softserial for the Split. Thats why I have RX in TX SS or is the split able to communicate bidirectional?

I don't think so.

> Could I use motor 5 for RX and motor 6 for TX?

Only if you are not using LED_STRIP.

jflyper, 2017-08-01:
@digaus The title is running SmartPort. So, we were talking about different situation all this time. I'm disappointed...

digaus, 2017-08-01:
Nono smartport is still the issue :) but led also

jflyper, 2017-08-01:
Disable your LED and run softserial over motor 5 and 6. Run whichever you like, SmartPort or RC Split, but not both.

digaus, 2017-08-01:
Kakute has a dedicated SmartPort pin which has a build in inverter. So I have SBus und UART3, SmartPort on UART1 and SmartAudio on UART6. For Split I now use SS on motor 5 and 6.

LED issue is related to SS nur SmartPort still does not work correctly only for short Time.

jflyper, 2017-08-01:
@digaus 

SmartPort: I suspect your SmartPort inverter is not functioning properly.

RC Split: It should work over software serial on motor 5 and 6 if LED and PPM are both disabled.

digaus, 2017-08-01:
Yes but that would mean that a lot of Kakute F4 are faulty if it is the inverter... 

Yes Split is working fine without LED.

jflyper, 2017-08-01:
> Yes but that would mean that a lot of Kakute F4 are faulty if it is the inverter... 

I have a F4 based FC with bidirectional inverter which does work stably for hours.
It must be your inverter or the receiver.

digaus, 2017-08-01:
Contacted Holybro for more Infos regarding a possible design flaw...  
Just one more question, could I use the RSSI pad (C01) for SS or for the LED?

jflyper, 2017-08-01:
> Just one more question, could I use the RSSI pad (C01) for SS or for the LED?

Unfortunately, no. It can not be associated with a valid timer channel.

[F405RG_PC0.pdf](https://github.com/betaflight/betaflight/files/1190268/F405RG_PC0.pdf)

jflyper, 2017-08-18:
@digaus Have the issue resolved?

closed 2017-08-19 by mikeller