betaflight/betaflight #3696: 3D mode  goes immediatly full throttle

X-FOOT, 2017-07-31:
i tested with Dshot 300  & multishot, before that, i first calibrated the escs in normal mode, thereafter set them to (reverse)bidirectional so that they spin correct
and enabled 3D mode
as soon as i arm the copter with the throttle stick in center and when i move the stick up and down the motors switch transition immediatly full throttle cw to ccw.

 i tested all kind of settings a whole day long, and it just doesnt work, my mate also tryed it and he has the same issue
the motors go immediatly full throttle when i move the stick away from the center
when i move it the other way they go full throttle in opposite direction
and i did everything as supposed, checked the rotation, enabled dshot , calibrate them!
However, i can hover inverted with the dshot reverse mode so i cant think what else it can be

KenImhof, 2017-08-01:
I am have the same problem with 3D mode transitioning from Normal to full throttle reversed.  Motors go directly from the 3D Deadband High (1550MS) to the minimum throttle (1500MS) just as the throttle stick goes slightly below center stick.

RipperDrone, 2017-08-15:
@X-FOOT @KenImhof Can you trace it back to the BF version or commit # where 3D started to be messed up? Think this might help the coders to narrow it down to the exact spot where changes have broken 3D functionality...

X-FOOT, 2017-08-15:
i only tested the new beta 3.2. i think its the first beta where it starts

KenImhof, 2017-08-15:
Problem started in 3.1.7.  Version 3.1.6 is the last working version and problem is not specific to DShot.

mikeller, 2017-08-16:
@X-FOOT: Is that only happening when you are using **reverse** 3D, or also when you are using normal 3D?

KenImhof, 2017-08-16:
Adding the two missing lines of code has fixed the problem.

if(feature(FEATURE_3D)) {
        if (!ARMING_FLAG(ARMED)) throttlePrevious = rxConfig()->midrc; // When disarmed set to mid_rc. It always results in positive direction after arming.

        if((rcCommand[THROTTLE] <= (rxConfig()->midrc - flight3DConfig()->deadband3d_throttle))) {
            motorOutputMax = deadbandMotor3dLow;
            motorOutputMin = motorOutputLow;
			throttlePrevious = rcCommand[THROTTLE];					//3D Mode Throttle Fix #3696
            throttle = rcCommand[THROTTLE] - rxConfig()->mincheck; //3D Mode Throttle Fix #3696
            currentThrottleInputRange = rcCommandThrottleRange3dLow;
            if(isMotorProtocolDshot()) mixerInversion = true;
        } else if(rcCommand[THROTTLE] >= (rxConfig()->midrc + flight3DConfig()->deadband3d_throttle)) {

X-FOOT, 2017-08-16:
so it is fixed now? where can i download the updated firmware to test it?
thanks!!

mikeller, 2017-08-16:
@KenImhof: Would you be willing to open a pull request for these changes?

KenImhof, 2017-08-16:
Pull request submitted.

RipperDrone, 2017-08-16:
PR does not show up in Pull Request section. Did I miss something?

KenImhof, 2017-08-16:
RipperDrone, please help.  I am sure I did not submit the pull request correctly.  I opened a pull request, added a comment and uploaded the modified file in a zip format.  I can see the request, but it is apparently in the wrong section per the attached picture.
[image]

RipperDrone, 2017-08-16:
Ouch, never done it myself either :-) Let's see...

RipperDrone, 2017-08-16:
Hm, looks like you just tried to

- direct PR to CleanFflight, not BetaFlight
- you sent in 18 (!) commits instead of just 1

RipperDrone, 2017-08-16:
Seems like you did not follow the expected route of how to create PRs for BF. And it's strange: Whenever you press the "New PR" button in BF, it shows me it wants to merge your commits into Cleanflight??

I guess we need some more experienced dev's help here :-(

RipperDrone, 2017-08-16:
here's the GitHub workflow - have you done these steps?
https://guides.github.com/introduction/flow/index.html

mikeller, 2017-08-16:
Happy to help. @RipperDrone is right, you should read and follow the steps in the introduction above. The one thing that's missing there is that at the very beginning, before you create a branch, you need to create a fork of the betaflight project into your own GitHub account, and then create the branch on that fork.

Let me know how you are going.

RipperDrone, 2017-08-16:
Sorry, saw this quite late. @KenImhof Are you sure that mincheck should be used in the expression for throttle? I think it's meaningless in DShot contect, so the expression should be different here...

just not enough of an expert to fully understand all the scaling and conversion in 3D mode for DShot... maybe it needs the actual 3D mode main author / dev to insert the correct expression for throttle here. Secondly, I'm not sure whether the previous throttle value needs to be set in this else/if section or not...

Apologies, seems we are getting stuck here :-/

KenImhof, 2017-08-16:
To me the code looks correct and working as designed with the two additional lines of code.  The code does not store the motor throttle with an offset so subtracting the mincheck value is correct. The throttle range is then used to determine forward or reverse direction.  The ThrottlePrevious is used so the motor transition does not occur until the throttle is outside the deadband.

DieHertz, 2017-08-16:
Your fork/branch probably originated at a commit in cleanflight rather than betaflight, but you can still change the destination when creating PR.

KenImhof, 2017-08-16:
I have already submitted a new PR request in betaflight.

Thanks,

X-FOOT, 2017-08-18:
i tested 3D mode in beta 3.1.6,   works perfect

leisim, 2017-09-15:
Is anyone working on this? I'd love to fly 3D but the lastest build still does not work :/

KenImhof, 2017-09-15:
Leisim,

If you want, I can compile the fix into a custom build for you.  What flight controller do you have?

You may also want to consider running iNav for 3D.  I am running a custom build of iNav for 3D with Compass/Barometer POS/ALT-Hold which is working for Inverted and normal bail-out as well as the standard RTH and Position hold.  I can post a video this weekend if you are interested.

RipperDrone, 2017-09-16:
Has your fix PR not been merged yet, @KenImhof? Thought this has happened quite a while back?

leisim, 2017-09-16:
@KenImhof wow that would be really cool. I have the KakuteF4. I never used iNav but I'm going to take a look now ;)
A video would be great but I don't want to waste your time ^^

KenImhof, 2017-09-16:
Just confirmed the pull request was implemented into 3.1.7.  If you are using DSHOT that may be the problem.  Looking at the code it looks like there may be a missing line of code.  

Video of iNav that was cut short during the RTH due to a problem with my phone is:
https://youtu.be/5oiWQhGGE78

leisim, 2017-09-16:
This looks so cool. Really nice flying ;)
I hope we can get it to work again. I'd try it myself but I'm new and don't know the BF code yet :/

KenImhof, 2017-09-16:
Make sure you are running version 3.1.7 and run ESC protocol Oneshot42.

leisim, 2017-09-16:
KakuteF4 does not support anything lower than 3.2 :'(

KenImhof, 2017-09-16:
[image]

KenImhof, 2017-09-16:
What ESC protocol are you running?

leisim, 2017-09-16:
Oh seems like it supports 3.1.7. I'm running Dshot600 atm but I'll try Oneshot42 and 3.1.7 tomorrow :)

KenImhof, 2017-10-08:
Incorporated Hard Deck and Rescue for similar 3D Reversing propellers into iNav3d.

Demo Video:
https://youtu.be/uNKoen8VKb4