betaflight/betaflight #3699: KISS CC board spins motors if powered up with arm switch turned on during power up

Tschingis, 2017-07-31:
Hi,

I have the following problem with my kiss cc driven copter when using latest bf3.2 builds:

reproduction:
- copter off, remote control off
- switch on remote control (taranis)
- turn on arm switch
- connect battery to copter
- copter initialises and motors start to spin

expected behaviour:
- copter initialises, motors do not spin
- disarm/rearm to start motors

I have this bug found in commit 33ac88b88d11d81923e7100d6a60106f018366de and most recent 94e1c9e76157d7327da52dd0f07e5c2e5f385c1f. I also tested another copter which has kiss fc which does not show the faulty behaviour. 

I made a default config where i only configured the arm switch so that it is easy to reproduce, here is my config:
`# dump
# Betaflight / KISSCC (KISS) 3.2.0 Jul 31 2017 / 18:04:21 (94e1c9e76)
name -
resource BEEPER 1 B13
resource MOTOR 1 B14
resource MOTOR 2 B00
resource MOTOR 3 B15
resource MOTOR 4 A08
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B03
resource PWM 1 A13
resource PWM 2 B03
resource PWM 3 A15
resource PWM 4 A02
resource PWM 5 B11
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 B03
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A13
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 B04
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
[... 310 lines cut ...]
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

# `

DanNixon, 2017-07-31:
Please provide a `diff` not a `dump`.

Tschingis, 2017-07-31:
`
# diff
# Betaflight / KISSCC (KISS) 3.2.0 Jul 31 2017 / 18:04:21 (94e1c9e76)

aux 0 0 0 1800 2100
profile 0

rateprofile 0

# `

DaEmpty, 2017-07-31:
I have 2 copters with BF 3.2 with this behavior (XRacer and KissCC). 
A third copter with Lux V2 Race and with a nightly build from last week does not have the problem.

# diff
# Betaflight / X_RACERSPI (XRC3) 3.2.0 Jul 31 2017 / 21:12:06 (dc06fc0d6)

feature -RX_PPM
feature RX_SERIAL
map TAER1234
serial 2 64 115200 57600 0 115200
aux 0 0 0 1375 2100
set serialrx_provider = SBUS
set min_throttle = 1050
profile 0

rateprofile 0

# diff
# Betaflight / KISSCC (KISS) 3.2.0 Jul 30 2017 / 19:44:04 (5b31ca291)

feature TELEMETRY
feature AIRMODE
feature DYNAMIC_FILTER
map TAER1234
serial 0 32 115200 57600 0 115200
aux 0 0 0 1725 2100
set gyro_sync_denom = 2
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_hardware = NONE
set motor_pwm_protocol = DSHOT600
set battery_meter = ADC
set pid_process_denom = 1
profile 0

set dterm_notch_hz = 0
set dterm_setpoint_weight = 151
rateprofile 0

set tpa_rate = 12

Tschingis, 2017-08-01:
Hi, I like to add some more information since i have taken a look at the target.h/c for kisscc and kissfc where only some hardware defines differ, which could not explain the different behaviour of kissfc and kisscc on the same sourcecode build. 

The kisscc has an X4RSB receiver and the kissfc has an XSR receiver attached. I do not have looked into the bootup sequence of betaflight but what me bothers is that the board with the XSR receiver does beep its motors only once, which shows esc is initialised but no signal, so its clear that during init the bf bootup sequence blocks. So this is my theory what happens (which lets fail the kisscc board):

on kisscc (X4RSB):
- battery connected
- fc board and x4rsb starts to boot
- x4rsb puts out some default sbus signal where channel of arming is low (off)
- bf reaches a point where it checks valid receiver signal ,and receiver signal has arming low
- signal is given, so bf continues to boot
- x4rsb starts to evaluate real received signal and changes sbus signal to real received signal
- bf gets real signal and sees, copter should be armed
- bf turns on motors

on kissfc (XSR):
- battery connected
- fc board and xsr starts to boot
- xsr puts out no pulses
- bf reaches a point where it checks valid receiver signal ,and receiver signal has arming low
- signal is not given, bf waits until signal condition is ok (this is why the motors only beep once)
- xsr puts out a valid signal which is also the received signal, with arming high
- bf waits, signal is now valid, but arming is still not low
- the user turns of the arming switch on the remote
- bf can now boot completly since arming is no longer high and signal is valid

I have written this scenario since i have seen the pullrequest 3706 and this requires an timeout, but it seems you have already edited your last comment before i started to write this :)