betaflight/betaflight #3701: BLheli passthrough doesn't work with BF 3.2.0 on Naze rev5

flint723, 2017-07-31:
When I try to connect with BLheli configurator I get this message:
"This firmware version does not support BLHeli passthrough. Please upgrade to a newer version using a FC Configurator specific to your firmware."

-Naze rev5 had 3.1.7, pass through works.
-I flashed 3.2.0 #24, #27, #37(the release candidate), none work.
-I flashed back to 3.1.7, it works.

Other than that 3.2.0 has been working well, flies great.  
Quad is using a Spektrum 4649t with bi-di srxl.  ESCs are Littlebee 20a (v14.85 multishot firmware).

Also, BLheli pass through is working fine with 3.2.0 on my PIKOBLX boards.

[Naze rev5 cli dump blheli passthrough not working.txt](https://github.com/betaflight/betaflight/files/1189064/Naze.rev5.cli.dump.blheli.passthrough.not.working.txt)

jflyper, 2017-08-01:
@flint723 BLHeli pass through has been dropped for 3.2 on NAZE target, because of the flash size constraints.

flint723, 2017-08-01:
Ok thanks.

closed 2017-08-01 by flint723