betaflight/betaflight #3702: BF3.2 Radiance no #1 motor spin

claytonbakerjr, 2017-07-31:
This morning I attempted to flash 3.2 onto my Furious Radiance board using this file: betaflight_3.2.0_FF_RADIANCE-37-dc06fc0d6.
Calibrated ACC and then Calibrated ESC's. All tones were normal during calibration. Even tones upon solid power on. In BF Configurator (I am running 3.1.3) I went to test each motor individually and #1 motor would not spin up yet the other 3 did fine. Opened BL Heli configurator and #1 esc did not show.
After inspecting the esc on #1 motor I did not find damage or signs of, this particular quad flew fine last night and experienced no crashes and was the last one on the field before leaving. So prior to checking\changing esc I decided to flash back to BF FW 3.1.7, repeated above procedure and all motors worked. Checked it in BL Heli and all esc's were present.
I reflashed the above 3.2 Radiance and it repeated same issue.
I then loaded betaflight_3.2.0_FF_PIKOBLX-37-dc06fc0d6 and repeated the above and at this point everything at face value is working.
The esc's I am currently using are Sunrise Cicada's 30amp loaded with AH-25 16.6 firmware.
I am not running DSHOT or trying to run DSHOT. 
Motors are EMAX RS2205-S 2600KV
Currently I have not loaded any type of config on the board since I am experimenting with other things and learning more about 3.2 so if more info is needed please advise

jflyper, 2017-08-01:
@claytonbakerjr FF_RADIANCE is not supported target of the BF3.2.

jflyper, 2017-08-01:
@MJ666 Any comments?

claytonbakerjr, 2017-08-01:
Thanks, since I am not that in depth with the programming I am not really sure what to reply with. So in the email I seen a reply that the Radiance is not a supported target, that being said will it be removed? Was the PIN swap meant to accommodate a DSHOT Config by chance? I will be moving forward with the PICOBLX target to determine if other items start to show themselves. Thanks Appreciate the response. As always guys keep up the good work.

jflyper, 2017-08-01:
@claytonbakerjr Sorry for the e-mail, which was based on the initial search for the target directory; turned out that the FF_RADIENCE is a sub-target of FF_PIKOBLX. So, it is supported.

claytonbakerjr, 2017-08-01:
@jflyper thanks for the clarification :)

MJ666, 2017-08-01:
Please see my comments at #3158. Likely it is the non DSHOT version of the Radiance and need an remap for PPM and MOTOR1.

`resource motor 1 A04`
`resource ppm A07`

For using this with DSHOT the ESC likely need to be connected to the PPM pin as recommended for the PikoBLX. Furious FPV will make an statement at there website wich target will be used for there individual FC's.

claytonbakerjr, 2017-08-01:
Thanks @MJ666

closed 2017-08-02 by jflyper