betaflight/betaflight #3710: 3.2 rc2, logging rate disabled, cannot change.

riquezjp, 2017-08-01:
Not sure if this is a configurator issue or rc2.
I flashed & set up for DYSF4 (using dysf4pro target), no problems. I set onboard flash & 1khz logging.
I went out flying, great - recorded BB data, no problems!
However, when downloading the data later I noticed the **logging rate drop down box is blank** & if i change it & save, it returns to blank  - when i click on the dropdown it has "disabled" ticked.

borisbstyle, 2017-08-01:
@basdelfos @martinbudden Any ideas?

riquezjp, 2017-08-01:
configurator 3.1.2 does not have this issue.

martinbudden, 2017-08-01:
@riquezjp , can you repro using "full chip erase" and see if problem persists?

@borisbstyle - perhaps we should have upped the MSP version.

riquezjp, 2017-08-01:
It does it on 2 quads. The other is using jenkins #1440
Using configurator 3.1.2 it works fine on both quads, so it must be the configurator?

martinbudden, 2017-08-01:
Did you do "full chip erase"?

riquezjp, 2017-08-01:
I have not since your comment no. not yet. but I did full chip erase when flashing this morning.

riquezjp, 2017-08-01:
re-flashed but the json restore doesnt work, so will set up again. 
At the moment the logging rate dropdown is working though.

DieHertz, 2017-08-01:
There was a related bug described #3712, now fixed.

closed 2018-04-21 by DieHertz