betaflight/betaflight #3712: Unable to disable Blackbox logging

wolverin3, 2017-08-01:
Hi guys,

I noticed this problem in RC2 but not in RC1.

I'm unable to disable Blackbox Logging for PikoBLX and Betaflight F3 hex.

I'm using BF Configurator 3.2.0.

Thank you.

DieHertz, 2017-08-01:
Enabling blackbox logging doesn't work as well. BlueJay F4 Mini, off by default, can't set to SPIFLASH, resets after reboot.
Trying to `set blackbox_device = spiflash` in CLI provides no echo-output and freezes the board.

Arakon, 2017-08-01:
I can disable logging for SPRF3EVO by using the CLI, but not the 3.2.0 configurator.

DieHertz, 2017-08-01:
There seem to be multiple issues with it, trying to investigate.

DieHertz, 2017-08-01:
Suspect this check to be the culprit:
```
// Don't allow config to be updated while Blackbox is logging
if (blackboxMayEditConfig()) {
```

Also found an inconsistency with `p_denom` value sending from BFC and reading in firmware.

DieHertz, 2017-08-01:
Fixed In #3715 and https://github.com/betaflight/betaflight-configurator/pull/

wolverin3, 2017-08-01:
Does this means that BF Configurator 3.2.1 will be out soon?

DieHertz, 2017-08-01:
Not necessarily, releasing a new version for each fix will take us far in terms of numbers. We have to wait a bit for fixes to accumulate.

closed 2017-08-02 by martinbudden