betaflight/betaflight #3713: receiver tab all values at 0

bentley91687, 2017-08-01:
hey im trying to connect a taranis qx7 using the x4r sb . i got the tx and rx binded properly and setup up serial and sbus rx in the port and config tab but when i go to the receiver tab all the values are zero instead of the 1500 they are usally. equipment is-betaflight f3,taranis qx7,x4r sb. any help with this issue would be great thanks

jflyper, 2017-08-01:
@bentley91687 Let us know your BF version and BFC version.

2black0, 2017-08-03:
the configurator is not suit with your BF version, for simple solution is, if you are using older BF firmware (ex : 3.0.1) you need run older configurator too, just install the older one

yeah, we know 3.2 is big different with before, so do the configurator

2black0, 2017-08-03:
[image]

I have 2 different version of Configurator, the latest (will update automatically while there is an update) and the older one

rboletto, 2017-08-11:
Hello, same issue, how can I download 2 configurators? I've tried to download again and came the 3.2 again, I I can't find older version.

rboletto, 2017-08-11:
Already tried older version and still same issue

mikeller, 2017-08-12:
@2black0: What you are seeing is most likely one configurator installed from the Chrome web store, and the second one that was installed from GitHub (locally).

2black0, 2017-08-12:
@rboletto : install older version here
https://github.com/betaflight/betaflight-configurator/releases

IndyBob2019, 2017-08-12:
How do you install older version? I downloaded the source where do I put it?

mikeller, 2017-08-13:
@IndyBob2019: https://github.com/betaflight/betaflight-configurator#alternative-way

rboletto, 2017-08-13:
Already installed older version and still zero on receiver page. I've reflashed the firmware and still same issue.

IndyBob2019, 2017-08-13:
Yeah I installed an older version of the configurator and same problem 0 on all tabs

IndyBob2019, 2017-08-13:
Should we try flashing an older version too? Like a couple versions back to be safe? I'll try that in a few.

Extent421, 2017-08-14:
the new configurator trashes portions of your configuration ( I think maybe by saving changes from the configuration tab?) once it's been broken going back to an old configurator won't help.  You need to go into your cli and check all of your min/mid/max usec values, min check, and rc interpolation at least.  Once those are reset to sane values the receiver tab will work correctly again.

mikeller, 2017-08-14:
Different theory: This could be caused by an invalid configuration being stored after using different RC versions. Try resetting the config to defaults, then restoring the settings from backup.

rboletto, 2017-08-14:
Problem solved. The miracle of firmware reflashing just happened. After several firmaware reflashing the receiver tab came back to normal on the latest configurador version (3.2....)

mikeller, 2017-08-14:
@rboletto: As I stated above, resetting the configuration to Defaults and restoring would have given you the same result, without trial and error.

redstork, 2017-08-15:
Hi I am having the same problem except mime was working properly and then was trying to calibrate esc and then it just disappeared.  Would this same process fix my problem  ?

Thanks for any help .

mikeller, 2017-08-15:
@redstork: Most likely yes. The configuration can get corrupted when switching between different RC versions. A reset to defaults will fix that.

closed 2017-10-22 by jflyper

rboletto, 2018-02-15:
Hello everybody, why they don’t fix this configurator issue?

jflyper, 2018-02-15:
@rboletto If you still have this issue, it is most likely your configuration problem. Please use other channels on the net for support.