betaflight/betaflight #3714: Gyro/accel noise while sitting on table

kylea19, 2017-08-01:
I just built up a quad with a new Betaflight F3 FC. When trying to tune it, there is a very noticeable high frequency vibration when flying, no matter what the maneuver (even just hovering). I spent days and many packs trying to tune it out to no avail.

   I searched around and found some videos on motor vibrations affecting the Accel/Gyros during flight, and even though I have brand new, never used motors (emax RS2306 2400KV) I thought I'd test it out. What I found was somewhat concerning, as my FC is showing that I have "massive" vibrations while not even spinning the motors. The picture below is a screen shot of the different axis' after sitting undisturbed on a table for about 10 seconds. I have the FC sandwiched between 2 rubber soft mounts on each corner, but I still get this phantom vibration.

[image]

Is this a filtering issue? I have a second identical board and it seems to do the same when sitting on a table. (Admittedly I didn't try very hard to secure the second board to anything and it may have moved while taking in the data) 

Any one have any ideas? I'm willing to try anything, I kinda just want to fly haha

Thanks for looking,

-Kyle

Arakon, 2017-08-01:
That's not massive vibration, that's tiny. You probably have some device (your PC, for example) gently vibrating the table. To give it some scale: Gently tapping a finger on my desk where the FC is sitting on a soft mousepad causes the values to jump to 12-18.

claytonbakerjr, 2017-08-01:
@kylea19 I tend to agree with @Arakon. If you are seeing something in flight then it is possible that you may need to filter some items. When you say "sandwiched" are you stating that you are using something like O Rings to soft mount the board? Are you also soft mounting the motors? I used to use O Rings but switched to the rubber dampeners. That made a big difference.  Joshua B has some good info out there about adjusting the filters, particularly the one having to do with PT1.

kylea19, 2017-08-01:
Hey guys, thanks for the quick replies. I agree its not a massive vibration (that's why I put it in quotes haha) I've watched a lot of Josh's videos and it seemed like he was getting values of less than .01 when he ran his motors up to full speed one at a time. My FC seems to be sensing values above 1 while just sitting there. Is it possible that the older version of clean flight he was using uses a different scale compared to betaflight 3.1? 

I ordered rubber isolation standoffs yesterday but for the time being until they arrive I put one above the FC between the nut that is threaded onto the standoff, and one below the FC that rests on the actual standoff. I'll check out some more of his vids on adjusting filters when I get home tonight. I just wanted to try and eliminate some other possibilities so I don't get started down a wild goose chase. 

I'll probably try to soft mount the motors next, thanks for the help!

Arakon, 2017-08-01:
Make sure he wasn't using an older version, which measured ACC and not gyro.. gyro shows A LOT higher values.

Chill1993, 2017-08-14:
Having the exact same problem, however my BFF3 board was already soft mounted using the rubber standoffs. Did you figure out what was causing the high frequency vibrations?

kylea19, 2017-08-14:
What I ended up doing was soft mounting the motors and swapping to a more balanced prop. The vibration seems to have gone away now.

Chill1993, 2017-08-14:
Could I assume that it may have been unbalanced/noisy motors causing the problem then?

closed 2018-04-21 by DieHertz

QUPS, 2018-08-04:
For anyone coming here, turn on your accelerometer and use that for measuring vibrations (accel instead of gyro in dropdown). Values should stay pretty much at 0 on a table. I have a couple motors going bad causing a lot of vibrations, you can hear and see it in my video footage. 

You can't really see the vibrations very well with the gyro sensor, the graph looks mostly like a flat line even at peak vibration. However with the accelerometer, it's very visible in both the XYZ values and the graph. 

[image]