betaflight/betaflight #3716: Kiss fc target esc telemetry should be on tx

Drewlrox, 2017-08-01:
Hello, some time ago somebody over at the slack channel created a version of 3.1.7 for me that had the esc telemetry set to read on tx rather then rx and it works fantastic with the existing telemetry pads on the kiss fc without needing to do any sort of re soldering. I think this should be either an option or the default for kiss fc as thats how people will have them wired, This will allow them to flash back and forth without needing to re solder each time.

lichtl, 2017-08-01:
Just use
`set esc_sensor_halfduplex = ON`

Drewlrox, 2017-08-02:
Are you positive that works? When asking the person if there is an easy way to change this they said it could only be done before compiling the firmware. Also the betaflight wiki recently linked to a newer guide on getting it working and it shows resoldering. If it was this simple why does the wiki tell people to resolder?

lichtl, 2017-08-02:
Yes. I've tested this some days ago. This is the minimal diff I've stored.

`
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR
serial 2 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
set motor_pwm_protocol = DSHOT600
set current_meter = ESC
set battery_meter = ESC
set esc_sensor_halfduplex = ON
`