betaflight/betaflight #3717: Blackbox log stopping suddenly, followed by awful oscillations, no control and full throttle

beeb, 2017-08-01:
Hey all,

I think this issue might not specifically be related to 3.2 because it happened yesterday on 3.1.7 too. My quad behaves sometimes very strangely when subjected to high throttle values or propwash. First, blackbox logging suddenly stops and then the quad goes into horrible oscillations during about 500ms to a second, time during which I have no control over the quad and the throttle seems to be high (maybe it holds its previous value I have no way of knowing). Then, I regain control and all is fine again. This seems to be related to PID somehow since lowering the values makes this happen less often, at least that's my impression.

But even with relatively low values (see below), this still happens sometimes as witnessed in the two videos that you'll be able to download, along with blackbox files, here (offset is inside the video filenames): https://www.dropbox.com/s/pmnilk7uc0s6wli/blackbox.zip?dl=0

[image]

In the videos, the oscillations happen directly after the blackbox log suddenly stops, around the 1min30 to 1min40 mark depending on which video you're looking at.

I checked that my CPU (F4) was not overloaded, by connecting both battery and USB, with props off, arming the quad and shaking it around while monitoring the CPU use in the configurator. The CPU load did not exceed 11% during this test, so I'm guessing that's not the issue. I also don't think this is failsafe related since my Taranis did not complain about lost telemetry and the failsafes are setup to have low throttle. I have air mode enabled in the modes tab if that's of any use.

Here's my diff: https://pastebin.com/ZyQzqGSs

Additional note: this happened on 3.1.7 yesterday too, and I had no Anti-gravity, no crash detection (obviously), no dynamic filter (obviously).

Any help would be greatly appreciated! Thanks

DieHertz, 2017-08-01:
Try disabling `sdcard_dma`

beeb, 2017-08-01:
Will try thanks! Can you enlighten me about what this command does?

DieHertz, 2017-08-01:
Disables use of DMA for blackbox writes, it will decrease maximum logging rate, but it seems like we still have an issue with SDCARD DMA clashing with something else, e.g. motor updates.

beeb, 2017-08-01:
Okay, good to know. This command allowed me to find this seemingly related issue : https://github.com/betaflight/betaflight/issues/2631

beeb, 2017-08-04:
Hello all, coming back to say that I flew 6 packs today without any problem occurring, thanks to `sdcard_dma` being disabled. Confirming this as a temporary fix.

C0br4, 2017-12-14:
I encountered this issue today with sdcarm_dma being off. I'm running an OmnibusF4V3 with 8k/8k and only 1kHz logging. [Blackbox and video on youtube](https://www.youtube.com/watch?v=TrPZQQtzBxk) (Stitched together, Blackbox explorer stopped rendering video when loggin cut off)

closed 2018-04-21 by DieHertz