betaflight/betaflight #3722: Suggestion : scale PIDs to cell count

ctzsnooze, 2017-08-02:
An idea was raised on rcgroups to reduce PIDs automatically when a higher cell count battery was connected.

Boris pointed to the existing code in mixer.c where the vbatCompensationFactor was called an then simply multiplied the mix (up when VBat fell).  See:

https://github.com/betaflight/betaflight/blob/master/src/main/flight/mixer.c#L600
https://github.com/betaflight/betaflight/blob/master/src/main/flight/mixer.c#L613

Perhaps the same could be done when cell count differed from the default cell count.  

For example, if a user tuned PIDs carefully for 4S batteries, they may find the quad behaves more normally (less overshoot and heat) if the PIDs are reduced by 20% (mix*⅘) when used with 5S batteries, and to get normal responsiveness with 3S, the PIDs might need to be increased by 30% ie mix*4/3

The user interface could be a CLI line:

CELL COUNT COMPENSATION   0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10

Default of 0 would mean no cell compensation.

Usage: if set to 4, PIDs would be scaled when a battery higher or lower in cell count than 4S is connected.

Might be best to ignore 1S since it is probably USB.

I don't know how to program this but it might be useful. A linear adjustment may not be quite right, but at least most quads would fly OK straight away on 5S and maybe 6S without a full re-tune.

SteveKfpv, 2017-08-14:
I love the idea of this. 
Just last weekend I had to set up two pid profiles, one for 4S one for 5S. I multiplied the pids from 4S as you wrote (seemed intuitive) and it flew pretty well with no adjustments. Maybe P was a tiny bit low (not quite linear as the heaver battery offset the extra rpm) but it would be cool to just be able to swap in whichever cell count you want.

mikeller, 2017-08-14:
Not sure if this 'one size fits all' solution with fixed scaling allows enough control.

How about storing a 'cell count' for each PID profile. Then, on initialisation, after the cell count has been evaluated, the following process would be run:

1. Check if the cell count configured for the current profile is equal to the actual cell count;
2. if not 1., check if there is a profile with a cell count equal to the actual cell count;
3. if a matching profile is found, switch to it.

This way, users can configure completely independent profiles for different cell counts.

Abancalari, 2017-11-20:
A follow up to this, why not just expand the functionality of VBAT PID compensation. Instead of all this messing around with cell counts, just tune the pids at a certain voltage with a certain pack, and then have the PIDs scaled based on the delta from that voltage? Tune on 4S (Nominally 14.8v), but when you put on a 5S (18.5) it scales by some percentage.

mikeller, 2017-11-20:
@Abancalari: I would be surprised if there was anything like a linear relation between voltage / cell count and PIDs, so 'scaling by percentage' will most likely not have the desired effect.

Abancalari, 2017-11-20:
Was just a shot in the dark really. Enough people have mentioned it's a simple ~20% +/- when switching cell counts and the current VBAT scaling relaxes 25% from a full to a dead pack. So I assumed if that was true, you could apply the same logic through a larger voltage range. Kiss's implementation was a bit more detailed if I recall. You would set the percentages for each term and maybe each axis? Something like that. I don't really know if it's worth making this super elaborate setup.

DEvil0000, 2017-11-22:
I would guess its not linear but maybe logarithmic or maybe some area like for thrust out of kv and prop size. But linear with something like a rate and expo could be a good first try. I would not try to scale with cell count - instead I would scale with voltage directly (and maybe with motor percentage?).

Vidalcris, 2018-05-17:
Awesome ! i hope someone will work on this :-)

tavdog, 2022-06-29:
Time to close?

closed 2022-06-29 by haslinghuis