betaflight/betaflight #3723: Attitude error during hard deceleration aka throttle blip, throttle cut

Saxin, 2017-08-02:
I'm using 3.2 RC2 (I've experienced same behavior on 3.1.6) on 2 quads, during hard deceleration from fast forward flight - attitude isn't maintained within reasonable limits, with pitch developing major error and rolls follows.

Steps to reproduce:
1. Install BF 3.2 RC2
2. Adjust PID's, I've tested both aggressive and relaxed PID tunes (log attached here is for the more relaxed case)
3. Adjust anti gravity gain to 4 (4000 in CLI).
4. Air mode is set as MODE and assigned to a switch.
Flight:
5. Perform fast forward flight @ WOT , sharply cut throttle ( Air mode engaged).
6. Remove hand from right stick. (mode 2; eliminate jitters / user error).
7. Observe attitude change in both pitch and roll axis. 
8. Analyse BB log and again observe attitude error in both pitch and roll axis.

[image]

In addition I've tried increasing AG gain up to 7 - without much difference.
In case of more aggressive PID tune and error is handles better yet it's still not "rock solid".

Full log attached.
[LOG00077.zip](https://github.com/betaflight/betaflight/files/1193834/LOG00077.zip)
Times: 
1. 00:26.285
2. 00:28.079
3. 00:36.520

Saxin, 2017-08-02:
Related to #3394

RipperDrone, 2017-08-02:
confirmed on 2 latest builds Omnibus F7 and F4, both BF 3.2. Hard throttle cuts cause nose pitch down and slight yaw right. Not pronounced on older copters (with both BF < 3.2 and weaker motors)

KenMurphy, 2017-08-04:
I have the same issue with an Omnibus F4. I have swapped motors and ESCs in an effort to fix the problem. It pitches forward and yaws on full throttle->low throttle drops. PID changes do not help.

Saxin, 2017-08-04:
@RipperDrone , @RipperDrone - please make effort to attach log files and provide times of occurrence so devs. can take a look , thanks guys.

RipperDrone, 2017-08-04:
Ack 🙂 weather is bad so logging will take some time.

Occurrence: Anytime, after every hard throttle cut, irrespective of flight manoeuvres of flight history...

Yamaford, 2017-08-05:
[image]
@Saxin your Pids are really low and I'm guessing you have a good bit of weight fwd. You can slowly drag the marker fwd from the point where you took the screenshot and see the P is trying to spin the motors up up front but not with enough athority to stop the dip. D is not helping at all. You also have your set point slider pretty low too. 
[image]
See the little wobble in the P atfter the nose does come back up? Not enough D. your lines are super clean. you have a ton of room for D. You can bring P up too but D for sure. 

Raise your gyro LPF and d term LPF up by 10 and the pids will react quicker too. great running machine you have there. 

I hope its just a bug in blackbox explorer but your motor protocol says PDM???

Yamaford, 2017-08-05:
One last thought. Look at your I. your PID is working hard the rear motors when you punch out, mostly the left rear, (that motor may be a little weak BTW). I term is winding up and staying there for a second until after you cut the throttle. Look at where it intersects P. It's doing exactly what you want it too but the P and D should have taken care of that already. I is slow to react but at low throttle with your PIDS its the only thing active so it takes a second to get you where you want to be. Raise your P and D.

Saxin, 2017-08-06:
Thanks, yes Mot. #3 saturates first, as was mentioned this log is with very mild tune, I have logs with much tighter tune - it's less pronounced but still there, something has changed from 3.1.6 ... not sure what, I'll re-tune more aggressively and upload the log. Thanks for looking.

lostcontrol, 2017-08-07:
I noticed something similar. I rebuilt a LS-210 with beaten up DYS SE2205 2550kV motors with an Omnibus F3 in order to test BF 3.2 and the dynamic notch filters.

I had directly setup anti gravity to 4 since that got ride of attitude changes on rapid throttle changes on other builds but here it is still quite massive. My P and D terms are high enough. I upped my I terms to 50-60 without much improvement. I have now set anti gravity to 6, still need to fly test it.

schemen, 2017-08-08:
I have experienced the same. I will also try anti-gravity of 5 and 6 out once I get cleaner weather

Saxin, 2017-08-08:
@schemen , @lostcontrol - please make effort to upload BB logs with times of occurrence, we should make it easier for devs to diagnose.

lostcontrol, 2017-08-09:
I managed to quickly fly 2 packs after work today. Anti-gravity at 6 definitely helped and after raising my I terms on pitch and yaw and also P a bit on pitch, I think the attitude is quite stable during quick throttle changes. I will try to fly more this week-end before making any conclusion.

RipperDrone, 2017-08-28:
Even at anti gravity set to 9000 and I term as high as 70, it's still there 😔. In all 3 quadcopters I have on BF 3.2. Was not noticeable in BF 3.1.7 before...

universe360, 2017-08-31:
I can confirm this.  Have dys omnibus F4 fc/spedix es30hv/2207 2450kv was stuck at 3.1.6 an never noticed this issue having to crank of the antigravity gains so high..I have the same exact issue after flashing 3.2 with the pitch dipping to the right and yawing. I tried jacking my up PIDS to .6 today an it worked a bit and its now pitching toward the back, I believe this is from it PIDS over compensating.  I'm fairly new I took everyone's advice on doing the motor/esc swap.  Still there. Back to 3.1.6 it is.. Hope we get this sorted..

airbornone, 2017-09-12:
i am experiencing the similar problems with my quad, if im under 80% throttle everything seems to be quit stable, however after full throttle punch the quad pitches forward with some roll with no stick inputs. i have played with the pitch iterm and roll iterm with out any effects. i have also wound the anti gravity out to 6.0 which shows 6000 in cli. this only seemed to become apparent after upgrading to 3.2 RC3, im now on RC4 with same issues.

tarekmaalouf, 2017-10-05:
I have to tell you that we are assembling 100 f4’s right now using the lumenier F4 aio and with 3.2 bf we are having a great deal of acc problems. Issues range from failing to level in calibration to the quads tilting during a standard hover procedure. I opened an issue about this as well under the luxf4 name but seems as though 3.2 might be having issues with this because I did  it have it in 3.1. Also, I increased the pid to 8khz and it looked like it fixed it but that was short lived.

ee0u30eb, 2018-04-10:
Has this issue been resolved? I am experiencing identical issues on Betaflight 3.3 with a Matek F405 FC. As per above I have tried all manner of PIDs and massive Anti Gravity but to no avail. I have posted elsewhere and uploaded the following blackbox and flight video: https://youtu.be/XyO19MVlrRk
Blackbox log which accompanies that video is attached. 
[LOG00008.zip](https://github.com/betaflight/betaflight/files/1896003/LOG00008.zip)

RipperDrone, 2018-05-10:
Still not fixed...

DmitryG, 2018-05-28:
Having same issue with SP Racing F3 on betaflight 3.3

closed 2018-07-04 by stale[bot]

Saxin, 2018-12-10:
Someone knows if this can be reopened for tracking purposes ?