betaflight/betaflight #3725: A generic serial OSD / VTX / CAM protocol - openTCO

fishpepper, 2017-08-02:
After having a chat with DieHertz and Azol (runcam) we were thinking it would be great to have an open documented and lightweight serial OSD / camera / vtx protocol.
As this will have to move a lot of data to the OSD this protocol has to be as lightweight as possible.
My suggestion is based on what I use on my tinyOSD project right now.

I would suggest the following:
https://github.com/fishpepper/openTCO

In order for this to work efficiently we will need some optimizations in the OSD code of betaflight, I am already working on that.
I am going to submit a PR for betaflight integration once a workable state is reached. Contributors are welcome!

This is meant as a draft and I am open for discussion :)

ledvinap, 2017-08-02:
@fishpepper : You should consider packing multiple commands into single frame - most screen updates are done as sequence of operations and overhead for single-command operation is not that important. This will allow more packet-level metainfo without significant overhead (larger CRC/reliable packet synchronization/...)

This can be done at command level (place deviceId in packet header and append commands (each having known/explicit length) into body.

Or it can be done on command level - have write-strings command that implements 'subcommands' ( `[<pos/text><pos/text>] / [<text><controlChar><text>]` something like  compressed https://en.wikipedia.org/wiki/ANSI_escape_code)

In future graphical+color OSD devices will appear (runcam split implementation should be quite trivial), so number of commands per update will grow considerably

ledvinap, 2017-08-02:
BTW: here is very interesting way to get constant-overhead packet markers (1 byte per packet + 1 byte per 255 non-zero bytes): http://www.stuartcheshire.org/papers/COBSforSIGCOMM/ . Command sequence can be always resynchronized and implementation overhead isn't that bad ...
(and the idea is so nice it should be implemented somewhere ;-) )

nathantsoi, 2017-08-02:
this sounds fantastic. i was also hoping we could wrap up off-board control and other features in a lightweight, well versioned protocol

i was playing around with doing this in protocol buffers a while ago in [protocol buffers)[https://github.com/nanopb/nanopb]

also, http://lcm-proj.github.io/ came up in discussion, idk how much we want to talk about transport vs. only doing serialization at this point

```
› cat protobufs/defs/msg*
/*
 * Authors:
 *  Nathan Tsoi 2016
 *
 * This is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This software is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this software.  If not, see <http://www.gnu.org/licenses/>.
 */

syntax = "proto2";

import "msg_api.proto";
import "msg_osd.proto";

message Message {
    enum Type {
        Api = 1;
        Osd = 2;
    }
    required uint32 utime = 1;
    oneof payload {
        MsgApi msg_api = 2;
        MsgOsd msg_osd = 3;
    }
}
/*
 * Authors:
 *  Nathan Tsoi 2016
 *
 * This is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This software is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this software.  If not, see <http://www.gnu.org/licenses/>.
 */

syntax = "proto2";

message MsgApi {
  optional uint32 major_version = 1;
  optional uint32 minor_version = 2;
  optional uint32 patch_version = 3;
}
/*
 * Authors:
 *  Nathan Tsoi 2016
 *
 * This is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This software is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this software.  If not, see <http://www.gnu.org/licenses/>.
 */

syntax = "proto2";

message MsgOsd {
    enum VideoSystem {
        AUTO = 0;
        PAL = 1;
        NTSC = 2;
    }
    optional VideoSystem video_system = 1;

    enum UnitType {
        METRIC = 0;
        IMPERIAL = 1;
    }
    optional UnitType units = 2;

    message Alarm {
        optional string name = 1 [default = ""];
        optional uint32 limit = 2;
    }
    repeated Alarm alarms = 3;

    message Element {
        optional string name = 1 [default = ""];
        optional uint32 position = 2;
    }
    repeated Element elments = 4;
}
```

fishpepper, 2017-08-04:
@ledvinap: 
Packing multiple requests into one packet would require more interaction with the osd code.
Right now I can not identify which requests belong together. 
The protocol idea looks nice, I will definetely have a look into that.

@nathan:
I have used protobuf and lcm at work a lot. I doubt this will be efficient enough at byte level.

I modified my suggestion a bit and implemented it on tinyOSD. Every osd call to write char/buf is directly sent to the osd via the proposed protocol. Looks like the bandwidth at 115200 baud is enough for an  usable osd updaterate :)
I will publish a protocol decoder once the protocol is mature enough.

The protocol is not fixed yet, I am still open to all suggestions. I just implemented it to have something to work with for now ;)

azolyoung, 2017-08-07:
@fishpepper 
I'm going to update our serial protocol for split,

digitalentity, 2017-08-07:
@fishpepper I suggest re-using CRC8/DVB-S2 which is fairly reliable and already implemented and used for CRSF protocol.

azolyoung, 2017-08-07:
sorry, wrong operation, please remove the above commit which have not type finished.

@fishpepper
I'm going to update our serial protocol for split, and I would like to know, so if using crc8, if I need update all screen data(in first time, or other case that need update full screen data), I should split the osd drawing command to multiple packets?
Is't possible using crc16? so the protocol will  be more flexible, in the case of only one byte increase

digitalentity, 2017-08-07:
@azolyoung our packets are fairly small, chance of corrupt bytes is also fairly slim, CRC16 is probably an overkill :thinking:

azolyoung, 2017-08-07:
@digitalentity 
OK! I'll see it tonight! Thank you!

fishpepper, 2017-08-07:
@digitalentity: thanks, I forgot to mention that, I am using this crc already (even though I am using my own faster version). I will add it to the document!
Thanks for the link to the uav interconnect protocol. It looks interesting as well! It looks like it would introduce a lot of additional bytes which I would like to circumvent.

@azolyoung : I think we should stick to crc8. I do not see why crc16 should make the protocol more flexible.
Regarding sending all bytes/splitting: The fill region command is suitable for clearing/filling regions.
Then the only thing that has to be sent are the updates. I will add the flll region code to my tinyosd pr now. You can have a look at that later tonight.

For the split protocol: As far as I understood azolyoung he will rework this as well, right?
As I have no camera stuff in my current build pipeline (only osd and vtx for now), I have no suggestions for that protocol yet. So I am fine with encapsulating the current protocol (but keeping it generic where possible).

I will come up with a VTX suggestion the next days.

azolyoung, 2017-08-08:
@fishpepper 

Yes, I'll rework the split protocol in recent days.
OK, I heard you say that(crc16), I'm no longer friends with him on the split protocol:)

And about the fill region code, yes, I've did it same with you(we've implement a clear region command on split), But when we draw a partial changes of screen, we use a 3 byte struct to describe which the new char in pos, the struct is look like:
{
uint8_t x,
uint8_t y,
uint8_t c
}
so if the changes is over, and take a look this example, there are have 30 cols and 16 rows in our OSD grid system, so if there is a full screen changes, I should generate a packet like the following:
<header:8byte><cmd:8byte><datalen:2byte>
<x:1byte:0><y:1byte:0><c:1byte:'x'>
<x:1byte:1><y:1byte:0><c:1byte:' '>
<x:1byte:2><y:1byte:0><c:1byte:'x'>
<x:1byte:3><y:1byte:0><c:1byte:'x'>
...
...
<x:1byte:29><y:1byte:15><c:1byte:' '>
<crc16:2byte><tail:1byte>
That is why I want to use crc16 on crc field. But If use crc8, I also can split this command to multiple packet if the packet length more than the 255 bytes

digitalentity, 2017-08-08:
@azolyoung are you saying that Split will be able to draw OSD on the screen? Wow!

digitalentity, 2017-08-08:
@fishpepper what are the changes to Betaflight's OSD code that are required for openTCO? I'm eager to make INAV compatible as well (also with tinyOSD)

azolyoung, 2017-08-08:
@digitalentity 

Yes, I'm adding a feature to make split to be a DisplayPort device, so it can used as a OSD device by FC.

fishpepper, 2017-08-08:
```
There are have 30 cols and 16 rows in our OSD grid system, take a look this example, if there is a full screen changes, I should generate a packet like the following:
<header:1byte><cmd:1byte><datalen:2bytes>
<x:1byte:0><y:1byte:0><c:1byte:'x'>
<x:1byte:1><y:1byte:0><c:1byte:' '>
<x:1byte:2><y:1byte:0><c:1byte:'x'>
<x:1byte:3><y:1byte:0><c:1byte:'x'>
```
I doubt you will get sufficient updaterates with this approach. Probably <8fps when you fully utilize the serialport at 115200 baud. This is why I put so much effort into this efficient protocol.
You can make your life much easier with my suggested changes to the bf osd code (partial updates).
There is no need to redraw the full screen. When you implement the following:
```
writeString(uint8_t x, uint8_t y, const char *buff); // buff is zero terminated
writeChar(uint8_t x, uint8_t y, uint8_t c); // single byte write
fillRegion(uint8_t x, uint8_t y, uint8_t width, uint8_t height, uint8_t value); // fill a region
clearScreen() = fillRegion(0,0,255,255,' ')
```
You are all set. The write String can use auto increment to draw consecutive chars (in x and y direction). 
My partial osd pr (based on martins suggestion) will only send the necessary updates
(= either drawing text or clearing the region). 

@ digitalentity: see the osd_device pr #3634 
Note that this is work in progress, you can give it a first try using a max7456 based board.
You will also need this bf configurator branch https://github.com/fishpepper/betaflight-configurator/tree/osd_device

digitalentity, 2017-08-08:
@fishpepper thanks, will have a look. OMG - the changes look massive, I need to have a carefull look at how to migrate them to INAV without breaking anything :smile:

fishpepper, 2017-08-08:
@digitalentity:
Yes, this will require a rework of the OSD system. But it removes the requirement of the large 1k ram framebuffer for the max7456 driver and will allow any osd device in the future. In principle you deal with four changes:
1) make osd code multi device
2) relative osd positioning
3) successive osd updates
4) adding opentco/tinyosd code

I will clean this up further, I am slow, I can only work on it in the evening hours.

azolyoung, 2017-08-08:
@fishpepper 

Haha, I think I've made a bad example, I just want to assume, in the worst case, if there are full screen or big data changes, we may need split it in multiple packet.
And in my test code, I've implement a partial drawing to draw the changes in the screen, it won't draw fullscreen data every time.

digitalentity, 2017-08-08:
@fishpepper I'll wait until you clean this up, then pull the code over. I'm not very fast as well :+1:

fishpepper, 2017-08-10:
I did some experiments with the new protocol, updaterate etc look really promising. I hacked together a short live preview of my new gyro fft spectrum overlay as well: https://goo.gl/oFUezE

DieHertz, 2017-08-10:
Your OSD is so pleasantly smooth

llambkin, 2017-09-13:
I cant wait to have a play around with this code, Hectech FPV is sending me a prototype VTX to have a play with. It should work with this project.

expipiplus1, 2019-03-08:
Did anything come of this?

Docteh, 2019-04-22:
Looks like fishpepper let his design loose on the world shortly after you asked. Someone would need to pick it up.

https://fishpepper.de/2019/03/11/tinyosd-tinyfinity-a-tiny-opensource-video-tx-with-full-graphic-osd/

closed 2026-09-26 by blckmn