betaflight/betaflight #3727: cli setting save error

robschii, 2017-08-02:
when i will save some settings in cli mode i get this error, also @ osd font upload

fc omnibusf4   flame f4 hlgrc

[image]

daro6, 2017-08-02:
I have the same issue with my Betaflight F4 (no SD).
Version: Betaflight / BETAFLIGHTF4 (BFF4) 3.2.0 Aug  2 2017 / 10:35:03 (11f1de561)

OSD does not show anything!

jflyper, 2017-08-03:
@robschii I can't reproduce the problem with OMNIBUS AIO F4 (1st gen); both CLI saving and font uploading.

 Which version of configurator are you using?

robschii, 2017-08-03:
I use the bf configuratior 3.2.0

robschii, 2017-08-03:
Some settings saved but when I copied some commands from text file it isn't saved. Specially profile rates

libwtf, 2017-08-03:
i see these errors alot too - but normally it saves the changes ok....

try uploading the fonts while lipo is plugged in, did the trick for me :-)

lumenier lux f4 osd / configurator 3.1.2-3.2 / different bf 3.2builds

closed 2018-04-21 by DieHertz