betaflight/betaflight #3731: BF Configuration 3.2 Problem

2black0, 2017-08-02:
I have a board Femto F3 (SPracing F3Evo / SPEV) in my emax babyhawk
flashed with BF 3.0.1

the last update of BF Configuration was 3.2, I dont mention it before
I need to connected with Configurator cause just install RX Flysky A8S (Ibus) so I set it with configurator and I have found 2 bugs : 
1. In tab Configuration, whatever I set serial protocol for receiver, even sbus, ibus, or anything, after save and reboot it always back to spektrum1024
2. Idk why motor run when I plugin the battery, even I dont do anything, just stay in receiver tab to check the signal

So I install older BF Configurator for setting BF 3.0.1 for the solution, and it works normally

2black0, 2017-08-03:
maybe for the solution (cause I think 3.2 is totally different with the old one), release configurator only for 3.2 and keep the older one, or need to push people to flash the latest betaflight firmware?

wind0r, 2017-08-03:
Releasing new apps isnt possible anymore so your Solution doesnt work.
But you can still download the older version here as a zip and install it.

2black0, 2017-08-03:
Thanks @wind0r
Yep, maybe just give notice in new apps or in github readme, what version of BF firmware is compatible with the apps, 

I think its bad for some people, they dont wanna upgrade the firmware (maybe just get best setup and no have problem with current firmware) but apps update automatically, and while do setting with the configurato current setting was broken (we dont know what will happen),

closed 2018-04-21 by DieHertz