betaflight/betaflight #3739: OSD bug with pre-arm + request

KenMurphy, 2017-08-03:
If the quad refuses to arm due to the pre-arm not being activated the OSD flashes 'load' as the reason for not arming. This is misleading, took me a bit of messing around to find the real reason after immediately forgetting I setup the new pre-arm feature.

Request:
If the pre-arm is activated and you arm then disarm, the quad won't re-arm until the pre-arm toggles states from inactive to active again. I would much prefer that if the pre-arm is in an active state the quad arms, no matter if the pre-arm switch was toggled or not. I setup pre-arm on a slider (above 75%) so that when I crash or am done flying I can turn the slider down to 0% and there is no chance of the quad arming. The problem is that with this setup even if I leave the slider in the 75%+ position the quad refuses to arm after a disarm until I drop the slider below and back above 75%. Meaning I can no longer re-arm midair easily if the crash allows me to. I disabled pre-arm due to this but I would like to use it for safety.

wind0r, 2017-08-03:
If i remember correctly the OSD 'load' issue should be fixed with the newest release and will be in rc3.

the request you have is something you should setup in your radio. you can mix the 2 switches to one there and do other fancy stuff

DanNixon, 2017-08-03:
The issue with the flag name was fixed in #3709

KenMurphy, 2017-08-03:
@DanNixon Thanks, I must have missed it. Any thoughts on the pre-arm request? I would like to avoid doing it in the radio for simplicity but if others feel differently then I guess I'll have to. I don't think I'm the only one that would like to be able to re-arm while using pre-arm though, especially when racing.

DanNixon, 2017-08-03:
Having prearm on a momentary switch seems the easiest solution if you don't want to do much configuration on the radio.

KenMurphy, 2017-08-03:
My momentary switch is the same across all my quads (voltage callout) so I would prefer not to change it.

DanNixon, 2017-08-03:
It can just as easily do both. (I'm in the same boat, my momentary switch does voltage and RSSI)

KenMurphy, 2017-08-03:
I know it can, its just not really the behavior I'm looking for. I basically want to be able to set my pre-arm once at the beginning of a pack and then not think about it until the end of the pack. Getting throttle to 0 and the arm switch back in position is enough to ensure that yes, I really do want to re-arm imo.