betaflight/betaflight #3740: Motors won't arm if attitude of quad is not flat

1banger, 2017-08-03:
Trying to launch at racing quad angles does not work. Just sits there and beeps. Once set back to a level
attitude, motors arm fine.

This is not an issue in 3.2.7. Have to have quad in a flat attitude to arm motors. I have attached a dump for more information.
[BTFL_cli_20170803_090809.txt](https://github.com/betaflight/betaflight/files/1197280/BTFL_cli_20170803_090809.txt)

Transam2k4, 2017-08-03:
Go to the cli and type:

get small_angle

If might be that the default small angle is less than what its sitting. I typically set my small_angle to 180 so that i can arm and try to get unstuck from a tree. type in the cli

set small_angle = 180
save

1banger, 2017-08-03:
Thank you. I'll give that a go and report back.

1banger, 2017-08-04:
That worked. Thanks for the help!

closed 2017-08-04 by 1banger