betaflight/betaflight #3746: Bug with DYSF4PRO target with 3.2 RC2

Fauropitotto, 2017-08-04:
Flashing to this removed access to the Battery Voltage and Current sensor panels on the configuration screen, and the battery tab causes the board to hang. The voltage displayed on the OSD shows 0.0V, with no activity on the current sensor as well.
 
Flashing the DYS F4 back to the 3.1.7 OmnibusF4 target brought back all of these features. 
 
Has anyone else seen this with the DYS F4 AIO (fc/pdb/osd boards?)

Fauropitotto, 2017-08-04:
Was this never addressed from #3475 several weeks ago?

jflyper, 2017-08-04:
@Fauropitotto Since there is nothing wrong with the 3.2-RC2 DYSF4PRO target and vbat/current meter and access to battery tab of BFC, I regard this as support request. Please redirect your problem to the thread in RCG. Thank you in advance.

[image]

[image]

jflyper, 2017-08-04:
@Fauropitotto Since you already have the problem statement in the RCG thread, I will close this for now.

closed 2017-08-04 by jflyper