betaflight/betaflight #3749: [FEATURE REQUEST] Cap the throttle value during failsafe stage 1 to 1/4 throttle max

jaapp, 2017-08-04:
We had a failsafe incident in our flying group. The quad was running betaflight 3.1.7 with the default 'failsafe' settings. Precisely during a high speed turn (at roughly max throttle), the flightcontroller went into failsafe stage 1 for a full second and kept the throttle at roughly the max value.

Within a second the quad traveled 60 meters, nearly missing one pilot at ~30cm distance at 120km/h (75mph), and crashing into an fortunately empty chair._*_

**My proposal is to cap the throttle at 1/4 maximum during stage 1 and make this the new default.** This is enough for most racing quads to hover (not lose altitude), and in most high-throttle failsafe scenario's this will lead to a much safer outcome.

The sole intended purpose of stage 1 is to keep the quad in the air long enough to hopefully  recover signal. There is no need to keep a high throttle during this guardtime-period. 

_*We have footage of this event that i'm willing to share privately_

rodgerramjet, 2017-08-05:
I'm sure a lot of acro pilots would disagree so if this is done ensure it's configurable.

Zappelphilipp, 2017-08-06:
I am not sure if this would be a better solution because its supposed to keep going exactly as the last input told it to to. Is 1 second really the BF default? This is way to long, especially for a racing drone imo because of obvious reasons. I would every racedrone pilot suggest to set the stage one to 100ms or 200ms max. This is like an eternity from technical point of view.

borisbstyle, 2017-08-07:
See PR https://github.com/betaflight/betaflight/pull/3778.

Lowering the throttle immediately to 1/4 throttle would be a bad idea as it would result in many crashes during short failsafe glitches, which can be unnoticaeble currently. Especially with low to the ground flying.

My PR lowers the default delay to 0,4sec, which makes it 0,6sec in total with hardcoded 0,2sec.

jaapp, 2017-08-07:
Thans @borisbstyle . This is a good solution. I'd prefer a even shorter default delay though. 0,2 sec, which makes it 0,4 sec in total. Or even less.. Flying at 120km/h there you'll be traveling a lot of ground in this duration.

> short failsafe glitches, which can be unnoticaeble currently
This should definitely be something that can't go unnoticed. Having a failsafe warning flash in the OSD for the remainder of the flight is a very good idea. Failsafes are no joke.

wind0r, 2017-08-07:
@jaapp We also would prefer a min_check that is not 1100. But way to many "issues" get created when reducing values to the bare minimum.
Also Betaflight is not only for Quads that Fly 120km/h. Default values should fit multply quad types not just the ones you find at races.

You still can change that value on your own and force people to set it up that way for Raceevents.

Maybe some utility to check those values would be nice for for Raceevent Organizers.
Similar to "real" Raceevents where vehicles get checked and "sealed".

richard-scott, 2017-08-08:
I like the min_check value needing to be changed as it forces you to read up on the FC.

That said, perhaps there be some kind of calibration of stick values added to the BFC. You could then calibrate your TX ranges like you do when connecting to a simulator?

wind0r, 2017-08-08:
Thats derailing this issue now :/ .. Calibration should be done on Radio side since else you will lose resolution! But adding such stuff to simulator would be realy nice!

libwtf, 2017-08-10:
@jaapp i **think this is the responsibility of the pilot to set the failsafe according to the scenario**.
i can see that on a highspeed-racequad 1s at full throttle can wreck quite some havoc, but for me as a freestyle pilot the scenario is completly different. i want my quad to continue its trajectorie when an object blocks my signal until it's past it, throttlechange would mess that up.

borisbstyle, 2017-08-10:
Closing the issue as pr #3778 finds a good middle way for proposed and previous scenario

borisbstyle, 2017-08-10:
@wind0r You gave me idea to reduce it finally :D
https://github.com/betaflight/betaflight/pull/3800

let's see how much complaints we get now

closed 2017-08-10 by borisbstyle

wind0r, 2017-08-10:
@borisbstyle i think 1050 will be fine.. i remember we set it to 1020 a year ago and issue tracker explod :D