betaflight/betaflight #3750: VelociDrone GPL code

ledvinap, 2017-08-04:
It seems that https://www.velocidrone.com/ is using betaflight code, but I can't find sources anywhere.
Backporting VelociDrone changes can improve BF/CF SITL ...

Can someone who owns VelociDrone request the sources?

fbohmann, 2017-08-04:
i bought velocidrone, what exactly shall i ask them?

fiveangle, 2017-08-04:
https://github.com/batcavegames/csharp_betaflight_port

ledvinap, 2017-08-05:
@fiveangle : Thanks, I overlooked that.

Their approach is quite minimalistic, I would expect them to plug in considerably larger (and isolated) part of betaflight ... 

Technically their code is breaking GPL  (all code must be GPL compatible for this kind of inclusion), but it doesn't seem that there is something interesting related to BF.

fishpepper, 2017-08-06:
I got this reply when asking if they ship it including betaflight code:
```

I think you need to study GPL a bit closer as I don't think you understand it - it is a complex license but it allows for the method of implementation we are using an we do not violate the license.

Just because we include betaflight does not mean we have to opensource our code. If our code could not run without betaflight (ie. betaflight was a hard requirement) then GPL would apply to our entire code base. 
However, we include our own custom written flight controller and the whole sim can run without Betaflight code used at all - this changes how GPL applies. 
If the user switches on the BetaFlight module then we run BetaFlight code as a loaded module and in this scenario the only code we have to opensource is the BetaFlight module, which we have done on our website as we are legally required to do so.
We went though Steam Greenlight and as a part of that Valve software who own Steam had to check our implementation was legal and did not violate GPL, we passed this review and we do not have to open source our code and neither do Valve. Which is why they do the review of all projects using GPL as the legalities of the implementation apply to them the same as they do to us.

```

Alright.... :-X

DieHertz, 2017-08-06:
Sound a bit like "f off"

digitalentity, 2017-08-06:
Well, they don't necesarily violate GPL. They opensourced Betaflight module (which they must), but the sim itself may fall under "System Libraries" definition - in such case it doesn't have to be opensource. This case was not tested in court AFAK.

DieHertz, 2017-08-06:
I don't really have problem with them, what makes me more suspicious is the Motolab F4 Target which gets clandestine releases right from manufacturer without providing the source code

digitalentity, 2017-08-06:
Yes, firmwares are a different beast - they are not splittable and in Betaflight's case should be 100% GPL and open-source.

DanNixon, 2017-08-06:
FWIW the information relating to Valve sounds fabricated.

Second hand experience tells me the review process is a simple check of does the game actually work, does it contain any questionable content (i.e. in game content, not source code) and it is actually a worthwhile game. I'm not even sure Valve deal in anything other than binaries.

robert-hoffmann, 2017-08-06:
I rather see you guys not hitting on velocidrone, and being a owner of it myself having the luxury of being able to do tweaks/tests in the Sim, than them just going ok ... don't like it, well, we'll just remove it.

I really like this Sim and wish more had interfaces that reflect what we will actually configure on our actual drones.

ledvinap, 2017-08-07:
IMO definition of module/plugin is much stricter in GPL:
https://www.gnu.org/licenses/gpl-faq.en.html#GPLInProprietarySystem
https://www.gnu.org/licenses/gpl-faq.en.html#GPLPlugins

It would be great to have VelociDrone supporting SITL Betaflight, running as separate process, with all associated goodies (configurator,blackbox, flight modes ...)
It would be border case from GPL standpoint, but it would bring considerable benefits to CF/BF community...

digitalentity, 2017-08-07:
@ledvinap the GNU's view of GPLv3 plugin/module definition was never tested in any legal cases. 

My view is that the simulator software itself is a "Major Component" for the plugin. By the word of GPLv3 Major Component is "a major essential component of the specific operating system (if any) on which the executable work runs". To my understanding if the software, covered by GPLv3, can not be run without something else to provide interface to user and hardware - this "something" is a Major Component which is not subject to GPL restrictions. This point of view, however, also was never tested in court.