betaflight/betaflight #3751: KISS fc with betaflight 3.2 motor 3, 4 sometimes don't spin

chowonpapa, 2017-08-04:
KISS fc  with betaflight 3.2(ESC kiss 24a) motor 3, 4 sometimes don't spin.
PWM UNsync option can not help this. 
However When I down grade to 3.17 It works perfect. 
Anyway I go back to 3.17 for now.

FrenzyTheory, 2017-08-04:
i got the same issue but with #1 and #2 with emax magnum

chowonpapa, 2017-08-05:
[image]

chowonpapa, 2017-08-05:
I use Dshot 600.

mikeller, 2017-08-30:
@jflyper: Could this be related to the discussion around how signal inversion is implemented (cf. #4000)?

jflyper, 2017-08-30:
@mikeller I don't think so by the following reasons.
1. Motor 4 is on normal (P) channel.
2. target.c for KISSFC hasn't been touched for 6 month.

mikeller, 2017-08-30:
Fair enough, just thought might be related since all motor outputs are inverted for KISSFC.

jflyper, 2017-08-30:
@FrenzyTheory EMAX Magnum is totally different FC. If you have the problem, please raise it as a separate issue following the issue template.

jflyper, 2017-08-30:
@chowonpapa Did you do full-erase?