betaflight/betaflight #3752: Matek F405 using BF 3.1.7 or 3.2 going crazy vibrate

zhangyu0615, 2017-08-04:
my quad hardware configuration:
FC:Matek F405 (with ICM20602 gyro IC)
ESC:Airbot 35A 6S with BLHeli_32
Motor:EMAX RS2306 2750KV
Receiver:Frsky XSR

BF Configuration:
PID&Notch: base on bf instruction, well tuned
ESC Protocol: DSHOT1200
Loop Time:8k/8k
Anti_gravity: Off
VBAT PID: Off

I tried both 3.2 or 3.1.7 version. the problem is my quad will going crazy vibrate when I was doing level flight. the vibration is like setting a very big P param to roll or pitch, the quad is trying to keep it's attitude(PID is still running), but using too much force. if i using 3.2 version and 32k gyro/16k pid, my quad will get into this status very quick, may be 10 or 20 seconds after armed, even I hold the quad and shake it will trigger the crazy vibration. after that, I change to 8k/8k under 3.2 version. the quad will fly for 1 or 2 minutes.

I also tried bf 3.1.7 using 32k/16k, the result is same to 3.2. when the quad going crazy, it's aways under smooth flight, no any flip or roll action. and the blackbox data will cut off before the quad going crazy everytime.

The most flyable configuration is using BF 3.1.7/double 8k. the quad can fly more than 3 minutes. I think the FC is not halted, because I can disarm the quad, that means the FC received data from receiver. 

Forgot my poor English, and thanks!

the-scriptinator, 2017-08-04:
Should check tips page here he http://www.mateksys.com/?portfolio=f405-osd and this bug: https://github.com/betaflight/betaflight/issues/2631

zhangyu0615, 2017-08-05:
thank u very much

blckmn, 2017-08-17:
If you try and turn off SDCARD DMA in CLI.

closed 2017-08-17 by blckmn