betaflight/betaflight #3754: USB beep value ignored - Betaflight F3 (3.2-RC)

spuder, 2017-08-04:
Disabling the USB beep on a betaflight F3 flight controller doesn't disable the beep when USB is plugged in on the bench

## BetaFlight-Configurator Version
`3.2.0`

## BetaFlight Firmware Version

`Betaflight / BETAFLIGHTF3 (BFF3) 3.2.0 Aug  1 2017 / 06:40:45 (869dd1f00)`

I've verified that `beeper -ON_USB` is set correctly in cli and configurator

```
beeper -GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper -DISARMING
beeper ARMING
beeper -ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper BLACKBOX_ERASE
```

[image]

Arakon, 2017-08-04:
Actually, it does work. The real issue is that your vbat reading is probably between 3 and 4.5v. Thus it doesn't detect USB, but thinks a 1S LiPo is connected. I have the same issue on two out of three micros, which pull the USB voltage down far enough for it to think it's a 1S LiPo. Check the battery tab, top right side, to see if it says 1S LiPo.

spuder, 2017-08-04:
I'm not seeing voltage detected anywhere:

[image]
[image]

swilkens, 2017-08-04:
Which isn't surprising, since there is no VBAT reading while powered over USB. Just when you power over the actual VBAT.

Disabling the USB beep feature removed most of the beeps for me, the three slow beeps after connecting for example - disappear. The beeps during the initial connection stay. Save / reboot beeps also stay.

spuder, 2017-08-04:
I also have SYSTEM_INIT disabled. I would expect it to not beep when I plug in USB.

swilkens, 2017-08-04:
SYSTEM_INIT is the three quick short beeps, those don't go away on USB if you just disable USB. So there seems to be a problem there, SYSTEM_INIT only goes away while SYSTEM_INIT is off. Regardless of USB. Meaning the SYSTEM_INIT beep is probably missing a check for the USB switch.

There is one BEEP the moment you plug the cable, I can't find which one that is and it's not controllable by any of the BEEP switches.

spuder, 2017-08-04:
I haven't ever heard 3 beeps, here is a video of the single beep on USB connect that I'm trying to get rid off

https://youtu.be/fbJiIC5ajak

Arakon, 2017-08-04:
> Which isn't surprising, since there is no VBAT reading while powered over USB. Just when you power over the actual VBAT.

Not true. Many of the smaller boards don't have the diode that prevents the FC from reading the battery voltage via the VBAT pin. All of my 20x20 and 16x16 boards read the USB voltage until they are connected to an actual LiPo.

swilkens, 2017-08-25:
Just to add, 3.2.0-RC3 still doesn't disable the SYSTEM_INIT beeps with -ON_USB. I have to manually disable SYSTEM_INIT to get rid of the SYSTEM_INIT beeps while on USB power with -ON_USB. Is this intended?

The one beep on FC reboot that @spuder shows in his youtube video I am also unable to disable.

The beeper configuration issues I had in combination with RC2 and configurator 3.2.0 seem to have gone away with 3.2.1 and firmware RC3:
https://github.com/betaflight/betaflight-configurator/issues/596