betaflight/betaflight #3757: Not getting any signal after plugging to Betaflight

Fincky, 2017-08-04:
Hi!
I'm getting this strange problem on one of my boards. It happened to one of my friends the other day and I didn't really know what to think of it until it happened to me today as well.
We are both running a micro brushless board the same one as in the Eachine Aurora90 for example. They use the Omnibus 3.1.6 firmware.
It used to work fine and after plugging it in recently to the Betaflight Chrome App, it seems like the board just won't receive any signal anymore.
I checked that the receiver was properly bound, put the original #Dump All code that I had saved when I got it, nothing helps!
So, since both of our boards stopped receiving signal after they have connected to the Betaflight App, I thought it might have something to do with it.
I know that this board is not using its own firmware and therefore I'm not sure I can get help for it but I thought I would at least let you guys know that there might be something going on.

jflyper, 2017-08-05:
@Fincky You didn't update your firmware, correct? What is the version of the BFC (Betaflight Configurator - what you call Betaflight App)?

Fincky, 2017-08-05:
I'm on Mac so I'm using the Betaflight Chrome App it's version 3.2

jflyper, 2017-08-05:
@Fincky Is the quad still flyable, apart from not able to see receiver signals in BFC?

Fincky, 2017-08-05:
@jflyper No it looks like it's really not getting any signal, it won't arm or anything.

jflyper, 2017-08-05:
The firmware is still 3.1.6 OMNIBUS target?

Fincky, 2017-08-05:
Yes

jflyper, 2017-08-05:
And you fed the result from earlier dump?

Fincky, 2017-08-05:
Yes

jflyper, 2017-08-05:
Did you type `save` at the end?

Fincky, 2017-08-05:
Yes, I did :)

Fincky, 2017-08-05:
That's what I thought originally too when I saw it happen to my friend but when I was able to reproduce it on mine, I realized it is probably something else and it might have to do with the configurator. The receiver is getting power, reacts to the radio turning on and off, everything is properly soldered. On top of that, in either case, it was working right before connecting to Betaflight so i'm really not sure.

Fincky, 2017-08-05:
@jflyper Ok I just resoldered everything. It was hardware. Thanks for your help :)

closed 2017-08-05 by Fincky

jflyper, 2017-08-05:
Happy props 👍