betaflight/betaflight #3760: OSD does not work with NTSC 16:9 cameras

spuder, 2017-08-05:
## Problem

Betaflight OSD display is absent if using a 16:9 aspect ratio camera like the Foxeer Monster V2 and the video format is set to NTSC. Auto and PAL work correctly

[image]

There have been reports of *flickering* with 16:9 cameras, but no reports of completely *absent* OSD with 16:9 cameras that I can find. 

- [rc groups thread](https://www.rcgroups.com/forums/showthread.php?2771562-Foxeer-Monster-HS1194-16-9-Widescreen-1200TVL-Camera)
- [ github bug](https://github.com/betaflight/betaflight/issues/2080). (The solution was to switch to NTSC. )
- [betaflight f3 wiki](https://github.com/betaflight/betaflight/wiki/Board---BETAFLIGHTF3#faq--known-issues) 16:9 flicker is known issue

## Work around

Set Video Format to 'Auto' or 'PAL'. 

[image]

[image]
[image]
[image]

## Device 

Foxeer Monster V2 (16:9)
Betaflight F3
Betaflight / BETAFLIGHTF3 (BFF3) 3.2.0 Aug  1 2017 / 06:40:45 (869dd1f00)
[cli dump](https://gist.githubusercontent.com/spuder/f4162313ecccecca7005373b429bfcc5/raw/f0ae131899f7870785a32ad3c2b8924a707fa996/expiramental-3.2.md)

jflyper, 2017-08-05:
All 16:9 cams have this issue, or only the Monster V2?

spuder, 2017-08-05:
I only have a Monster V2 (16:9) and an Arrow V3 (4:3) to test with.

jflyper, 2017-08-05:
Can I change the title to "Monster V2 (16:9) not compatible with BFOSD" and add "Informational" label?

Arakon, 2017-08-05:
Are you actually sure that the camera is outputting a NTSC signal? Apparently Auto seems to detect a PAL signal afterall. If it was NTSC, the OSD text would be positioned higher due to the difference in resolution.

jflyper, 2017-08-05:
With (presumably) so many Monsters sold, I wonder why no body has scoped the output from them...

alfag4, 2017-08-05:
I think that there is confusion. Some say that have flicker problem in PAL other in NTSC and are not able record the OSD (PAL/NTSC) with their dvr.
I have a Monster V2 in PAL and NEVER have osd flicker and i'm also able to record the OSD using my eachine vr d2 pro DVR.
Today i will try to test with ntsc signal.

https://www.rcgroups.com/forums/showpost.php?p=37858625&postcount=2314

alfag4, 2017-08-05:
OmnibusF4 Pro
VTX Aomway TX001

Foxeer Monster v2 bought in june 2017.
I can confirm that the OSD is visible in NTSC and PAL.
The first photo is PAL the sencond NTSC.
Sorry for the bad quality.

[image]
[image]

spuder, 2017-08-05:
@alfag4 what version of betaflight? I suspect this bug is isolated to the F3 flight controller, or a specific betaflight version.

alfag4, 2017-08-05:
I think that depends by the FC. In various threads i read of many problems for this cam but honestly i have not one. The only issue is related to WDR setting.
Actually the 3.2 but i'm sure that the 3.1.7 also works because i'm using the NTSC for other test in the past.

jflyper, 2017-08-05:
Sync handling of the 7456 driver hasn't changed since last November.

And this is not a bug. This is a compatibility problem between the cam and the 7456 chip.

alfag4, 2017-08-05:
The OmnibusF4 Pro use the same IC of BFF3 so because i'm see the OSD in NTSC and spuder not. :-)

spuder, 2017-08-07:
I rolled back to 3.1.7 and found the same behavior. 16:9 cameras do not work if explicitly set to NTSC (auto works fine with NTSC cameras)

Other work is being done to change the 7456 dependencies https://github.com/betaflight/betaflight/pull/3417

jflyper, 2017-08-07:
> I rolled back to 3.1.7 and found the same behavior. 16:9 cameras do not work if explicitly set to NTSC (auto works fine with NTSC cameras)

Yes. The AUTO mode tries it's best to understand the incoming signal, and does it dynamically.

> Other work is being done to change the 7456 dependencies here #3417

That work has nothing to do with the camera incompatibility problem we are dealing with here.

fishpepper, 2017-08-11:
>AFAIK, all FCs with 7456 chip use the exact same circuit around the chip; the "typical application" given in the MAX7456 data sheet.

There are some using the original MAX7456 chip from maxim and some that use a clone (?) called AT7456. They seem to be compatible but might behave differently?
At least there are some reports on that: https://github.com/PlayUAV/PlayuavOSD/issues/19

Could you post what is written on the osd chip (the long ic with pins on only two sides, either labeled MAX7456 or AT7456).

Anyway this is nothing bf firmware can fix, looks like a signal incompatibility to me as well...

robert-hoffmann, 2017-08-16:
I have an eagle2 16/9 which i can test tonight on bf 3.2 rc3

spuder, 2017-09-08:
Closing 

I assumed that because the foxeer monster packaging said NTSC, the camera was running NTSC. 
On closer investigation, the camera was actually set to PAL. 🙁

closed 2017-09-08 by spuder

importon, 2017-10-27:
My GUI does not have the option to change to auto or pal.  How are you guys doing this?  
[image]

jflyper, 2017-10-27:
@importon 

You don't have the "Video Format" section, because whatever your board is, it's flashed with SPRACINGF3 target firmware, which does not know how to control onboard OSD chip.

It is not possible (at least for now), to control video format of serially attached external OSD devices; you have to change it according to the procedure of each device.

whiteflash444, 2017-12-15:
I have a diatone tyrant S 215, I bought a foxeer monster V2 and installed it, the camera is mostly excellent except my OSD is flickering and sometimes the video will just go black, does anyone have a similar issue?  Setting betaflight and the drone to NTSC didnt do anything....video link attached, wiring attached (i know its a bit messy) any help please? its appreciated.
[image]
https://www.youtube.com/watch?v=cHjK1dmJkYA

spuder, 2017-12-15:
whiteflash444 This sounds like an unrelated issue. It should be discussed in the slack channel, or a separate github issue.

whiteflash444, 2017-12-15:
"OSD does not work with NTSC 16:9 cameras" thats the title of the thread....thats the camera imposting about, how is this not related??

spuder, 2017-12-15:
This issue is closed. It was a misconfiguration in betaflight where the OSD wouldn't show up. The issue you are describing:

>flickering and sometimes the video will just go black

is not related to the OSD.