betaflight/betaflight #3762: BF 3.2 KOMB ESC - 1 not spinning

4cellOstrich, 2017-08-06:
I just flashed to BTFL 3.2 on 2 of my kombini quads, however, motor 1 won't spin on either.

4cellOstrich, 2017-08-06:
Note, they worked and work fine on previous FW versions.

jflyper, 2017-08-06:
Is that with shot? How about other RSV protocols?

jflyper, 2017-08-06:
I meant ESC  protocols.

MJ666, 2017-08-06:
Likely is an Kombini V1 (non DSHOT) and needs an remap of the motor port.

4cellOstrich, 2017-08-06:
Yep,  these are both v1 Kombinis running multishot.

4cellOstrich, 2017-08-06:
and I tried oneshot125 as well. Same thing.

MJ666, 2017-08-06:
Have you done the remap for the motor 1 port? More detail over here: https://github.com/betaflight/betaflight/pull/3158

4cellOstrich, 2017-08-06:
I looked it over but could not discern what to do.

4cellOstrich, 2017-08-06:
Do I just enter:
resource motor 1 A04
resource ppm A07

into the cli and save?

MJ666, 2017-08-07:
Exactly this should do it.

4cellOstrich, 2017-08-09:
It worked, thank you!

closed 2017-08-09 by 4cellOstrich

wolverin3, 2017-08-10:
This is for Kombini FC (Non-DShot version)
1. Flash FF_Kombini, BF 3.2 RC3
2. Bridge Motor 1 pin to PPM pin
3. DShot is now working.
No need for any resource mapping. Hope it helps. Cheers.