betaflight/betaflight #3763: rateprofile not working correct on betaflight f3

Tbeal79, 2017-08-06:
when in the pid, and you set your pid, rates or anything and you save for rateprofile 1, then switch to profile 2 it just flashes and puts it back to 1, but up on the cli it tells you that you switched even tho you have not. and it will do this for all 3.
this is also happing to the omnibuse f4 pro also, im going to try and delete the program and redown load it and see if that works. i hope it does.

chime13, 2017-08-06:
@Tbeal79, yes this is a confirmed issue that "I guess" will be resolved when @basdelfos gets back from holiday.
In the meantime, I suggest going back to 3.1.1, before the issue was introduced - That's what I did and it works again. HTH

Tbeal79, 2017-08-06:
thank you so much, i didnt even dload it, some how it went and did it for its self and i went in to try and change them and was like wtf? lol, then i looked and i seen it was bf 3.2.... i was like how did it do that when i didnt even change over to it bc i was going to wait till it was released. so all i have to to is go in and flash the 3.1.1 and it will work? im kinda a noob when it comes to this stuff hahaha

chime13, 2017-08-06:
You don't have to re-flash - it's not the firmware, it's the configurator. The firmware on the FC (3.1.5, 3.1.6, 3.1.7) is not the issue.
You'll have to manually dl the extension and load it up in chrome.
https://github.com/betaflight/betaflight-configurator/releases/tag/v3.1.1
HTH

mlametta, 2017-08-11:
Probably a known issue regarding this as well but the configurator does not show different rate profiles being loaded in the log area when switched to on the transmitter.  FC beeps though so it seems to be working fine otherwise.

basdelfos, 2017-08-30:
@Tbeal79 can you confirm this is fixed now. It is fixed in configurator PR https://github.com/betaflight/betaflight-configurator/pull/593 which already is merged in the latest BF configurator in the Chrome Store.

mikeller, 2017-08-30:
@basdelfos: Just gave it a quick check, works fine now!

closed 2017-08-30 by martinbudden