betaflight/betaflight #3765: Strange quad behaviour during flip (v3.1.7)

rtlopez, 2017-08-06:
Hi Guys,

I'm beginner in fpv and I have flown a little this weekend. I'm not using air mode yet and still and have motor_stop enabled in my quad and I'm aware of that it may cause . Once I tried pitch flip, lost control in the air and crash my quad. But I noticed something strange in quad behaviour, so I checked the blackbox log. The log confirms that I cut a throttle too much, but I observed that while I do the flip the front motors (1,3) unexpectedly started and revert my flip for a moment, and the continued flip. This cased me panic and you know what happened later ;) Are you able to determine the reason of this behaviour? do you thin it is a bug? See my attachments below:

[image]

- [blackbox log](https://drive.google.com/open?id=0B59moTTmxRJoYy1WWFNnOWFTeGc)
- [short dvr recording with overlay](https://drive.google.com/open?id=0B59moTTmxRJoNl9QLUVrdHBlcnM)
- [cli-diff](https://drive.google.com/open?id=0B59moTTmxRJoS2tEM2dURFp1Vk0)

I'm just curious what happened, particularly why motors started. Technical terms appreciated as I'm programmer too ;)

Saxin, 2017-08-06:
@rtlopez 
Please use air mode, there was a bug in 3.1.7 where quad mis behaves without airmode, issue #3016 

You always want airmode for low throttle extreme maneuvers - this provides full authority regardless of throttle value.

adrianmiriuta, 2017-08-07:
are you flying HEADFREE enabled ?

rtlopez, 2017-08-07:
Thanks for quick reply

@Saxin I've analysed a mixer.c code a little and my attention has attracted a fragment
`if (feature(FEATURE_MOTOR_STOP) && ARMING_FLAG(ARMED) && !feature(FEATURE_3D) && !isAirmodeActive()) {
            if (((rcData[THROTTLE]) < rxConfig()->mincheck)) {
                motorOutput = disarmMotorOutput;
            }
}`
despite of situation this fragment should keep motors off. But there are come coincidences.
1. previously I've experiment with auto-land failsafe procedure and set some parameters as follows.
`set failsafe_delay = 4
set failsafe_off_delay = 50
set failsafe_throttle = 1400
set failsafe_throttle_low_delay = 60
set failsafe_procedure = AUTO-LAND`
2. failsafeApplyControlInput() sets rcData[THROTTLE] to failsafe_throttle
3. motors started exactly 0.4 seconds after I cut the throttle as it is set in failsafe_delay

So I think in this situation the failsafe procedure has started and failsafeApplyControlInput() has been called for some reason, but it is not shown in blackbox log (failsafe is still idle)

@adrianmiriuta no, I'm not using HEADFREE.

Cherrs :)