betaflight/betaflight #3768: HOTT telemetry broken in 3.2.0 RC2

nhadrian, 2017-08-06:
Hi all, I have a small issue.
After updating to 3.2.0 RC2 and reconfiguring board from scratch, the HOTT telemetry looks like not working, the receiver doesn't recognize any connected sensors. Please note that hott worked in 3.1.7 with the same configuration (telemetry is enabled in config panel and HOTT is connected to UART3 with proper diode, ports panel also configured on the right way), hardware didn't change at all.
If any further information needed, just let me know.
Regards Adrian

TheIsotopes, 2017-08-06:
Since #3145 that has not interested anyone :((

nhadrian, 2017-08-06:
Oh sorry, I missed #3145.

closed 2017-08-06 by nhadrian

nhadrian, 2017-08-06:
after reading #3145 I tried 'tlm_halfduplex = OFF' setting and HOTT works!

Paradox32b, 2017-08-15:
Yes it's a shame...