betaflight/betaflight #3770: DYS F4 Mah/VBAT not reading

Kineticheli, 2017-08-06:
I have a DYS F4 board running BF 3.1.6. When I first got it flyng everything worked fine for the first half dozen flights. Now the Mah current just starts climbing when I first plug in and VBAT reads very low. I think it started when the GUI got updated, but can't be sure. Just wanted to see if it software bug or defective hardware.

Thanks.

AndersHoglund, 2017-08-06:
My DYSF4PRO shows reasonable values. Same values in GUI, on OSD and via telemetry on Tx (Spektrum SRXL).
[image]

jflyper, 2017-08-07:
@Kineticheli Does voltage and current displayed by OSD behave the same way?

Kineticheli, 2017-08-07:
> @Kineticheli Does voltage and current displayed by OSD behave the same way?

Yes, the OSD is displaying the erratic voltage and current readings. I guess it's time to try 3.2

jflyper, 2017-08-07:
Oh... That's a must...

Kineticheli, 2017-08-08:
Flashed to 3.2. I can access battery and power tab now. The current draw and voltage are still off. I see if I change the scale i can get it closer with the voltage, but even with the current scale maxed at 1500 it shows a draw of 7 amps. It all worked fine when I first got the board, could it be defective now?

scottinchina, 2017-08-08:
I have the same issue with DYS FC and firmware 3.1.6 , 3.1.7 and 3.2.0 , I still don`t know why the vbat lower than main battery volt, I am using vbat check from camera and display through osd.

jflyper, 2017-08-08:
@scottinchina How far off is your BF provided battery voltage reading from what ever you are using to measure it?

scottinchina, 2017-08-09:
@jflyper 
The vbat from BF3.1.6 using onboard ADC settings, the main battery voltage value after scale up to max 255 is between 3.2-3.4v. The method getting vbat direct from PDB to camera than showing to OSD is accurated.

Kineticheli, 2017-08-09:
I was able to adjust the scaling to get the vbat reading correct. I still can't figure why the current meter is way off. When I plug in it shows it is pulling 23.3 amps just sitting there. Its like the decimal is in the wrong spot.

jflyper, 2017-08-09:
@Kineticheli @scottinchina I think you two have a dud board. Please consult your supplier.

Kineticheli, 2017-08-09:
Test flew today, VBAT still going crazy even after adjusting the scale. I think it is time to just admit it might be a bad board. To bad, I liked the layout.

mikeller, 2017-08-10:
@Kineticheli: Just ask your supplier for a replacement board. Most of them are pretty good about replacing faulty boards.

closed 2017-08-10 by jflyper

Kineticheli, 2017-08-10:
I tried the set to defaults and the go strait to power/battery tab, it didn't work. I'll try again over the weekend when I have more time.