betaflight/betaflight #3772: Can't connect transmitter on Betaflight 3.2.0-RC3

Fresh88, 2017-08-06:
I can't connect any transmitter on Betaflight 3.2.0-RC3. I use Omnibis F7, i can't enable any serial port, it is always reset. But receiver and transmitter are connected (green LED on receiver).

jflyper, 2017-08-07:
@Fresh88 

(1) What is your BFC version?

(2) Did you do full-erase or cli `defaults`?

borisbstyle, 2017-08-07:
Did it also work on RC2?

Fresh88, 2017-08-07:
I use 3.2.0. I've done full chip erase during firmware updgrade. Before that i used 1467 firmware (http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1467/), transmitter has worked, but some functions in configuration settings couldn't be changerd (they appear to remain the same after save and reboot e.g. i couldn's change beeper configuration at all, all sliders are always enabled, also i couldn't change "motor direction is reversed" in mixer, it is always enabled).

Fresh88, 2017-08-07:
How to make CLI defaults?

jflyper, 2017-08-07:
Type "default" into CLI.

Fresh88, 2017-08-07:
Also, restore defaults function doesn't work at all.

AndersHoglund, 2017-08-07:
Some of the symptoms you are describing suggests that you might be using a too old BFC version, with known compatibility issues. Please try again with an up-to-date BFC application, currently version 3.2.1

richard-scott, 2017-08-07:
It would be great if the BFC had a built in list list of compatible firmware version numbers that it could work with and then notified you of the requirement to upgrade if the FC had a firmware that it could not work with. I feel that this would save lots of chatter asking for help.

I did open a feature request for this idea to be include in the BFC but they were not interested.

jflyper, 2017-08-09:
@Fresh88 Do you still have this issue?

jflyper, 2017-08-09:
@Fresh88 
I have tested the 3.2.0-RC3 OMNIBUSF7 target with SBUS receiver, and was able to use all hardware UART ports (RX1, RX3, RX6 and RC == RX2) for serial RX.

Fresh88, 2017-08-12:
I have changed the UART port ann now it works.

closed 2017-08-12 by martinbudden