betaflight/betaflight #3773: betaflight 3.2.1 not showing OSD FRSKY F3 board

applewood, 2017-08-06:
I'm using the new FRSKY XSRF3P0 board, with firmware version
# Betaflight / FRSKYF3X 3.2.0 Jul 14 2017 / 11:40:09 (a5f7fecc5)

I've noticed today that when using betaflight configurator today the gyro icon has greyed out, even though the gyro seems to be functioning according to the sensor page.  Also when selecting OSD fromn the menu, the OSD screen is blank.  This used to work, i'm sure its since powering the board by the lipo today.

I've tried reflashing firmware and restoring factory defaults, but not getting the OSD to function.

H/W fault, or am I doing something wrong.  Any help would be appreciated.  I've attached the dump output if it helps.
 dump
# Betaflight / FRSKYF3X 3.2.0 Jul 14 2017 / 11:40:09 (a5f7fecc5)
name -
resource BEEPER 1 C15
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 B01
resource MOTOR 4 A01
resource MOTOR 5 A02
resource MOTOR 6 A03
resource MOTOR 7 A00
resource MOTOR 8 B00
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
[... 460 lines cut ...]
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

jflyper, 2017-08-07:
@applewood Do cli "defaults" and check if it changes anything.

applewood, 2017-08-07:
@jflyper thanks for your help.  Ive tried that, unfortunatly the gyro is still greyed out and the osd is blank.

Would you have any other suggestions.

Cheers

jflyper, 2017-08-07:
@applewood Does LED flashes in some definitive pattern at boot, such as 5 fast flashes followed by 2 slow flashes, repeating this for twice?

applewood, 2017-08-07:
@jflyper, yes it does.   The blue light does 4/5 really rapid flashes, followed by one slow blue, followed 4 medium flashes.

applewood, 2017-08-07:
@jflyper , I've done a video of the boot sequence.   5 quick flashes followed by 1 long flash, followed by 3 medium flashes.  Can't seem to attach video here.

applewood, 2017-08-07:
@jflyper, yes it does. It does 5 rapid flashes followed by 1 long flash, followed by 3 medium flashes ..  so according to the FAQ
1 FAILURE_DEVELOPER: External interrupt of sensor failed to initialize.
2 FAILURE_MISSING_ACC: Accelerometer/gyro sensor is missing
3 FAILURE_ACC_INIT: Accelerometer/gyro sensor failed to initialize
4 FAILURE_ACC_INCOMPATIBLE: The found accelerometer/gyro sensor is not compatible/not the expected one
5 FAILURE_INVALID_EEPROM_CONTENTS: EEPROM/FLASH configuration content is invalid
6 FAILURE_FLASH_WRITE_FAILED: Write of configuration to EEPROM/FLASH failed
7 FAILURE_GYRO_INIT_FAILED: Gyro initialization of SPI MPU6000 accelerometer/gyro failed
The most common one seem to be error 2 where the accelerometer/gyro sensor can

So assume it's failure developer, but not sure what that means to my setup.  Not sure what the last three flashes mean ??

jflyper, 2017-08-07:
@applewood Does the 5-1-3 flash repeat several times? Or just once?

jflyper, 2017-08-07:
@applewood Wait.
Where did you get the FRSKYF3X firmware? It is not what Betaflight is officially providing. If the firmware is provided by and pre-flashed by the board vendor, you have to consult them, not us.

applewood, 2017-08-08:
@jflyper, the 5-3-1 just appears once, it's not repeated.   After boot the blue light then responds to movement.

The firmware was provided by FrSky, is there another version I could try?

Thanks for your help

applewood, 2017-08-08:
The board is xsrf3po if it helps

jflyper, 2017-08-08:
@applewood 

There's nothing we can do for you here, because we don't know the nature of the FRSKYF3X firmware your board is flashed with.

The closest BF repo has is FRSKYF3, although the board wiki says it has integrated X8R. You may want to give it a try.

Please note that flashing with FRSKYF3 firmware is one way, unless you have the FRSKYF3X firmware hex with you.

applewood, 2017-08-08:
@jflyper I'll happily try the FRSKYF3 firmware but I can't see it on the board types in betaflight configurator?   I can see it listed as a board in GitHub.  i'll have a look around to find it.

The firmware from FRSKY is as follows, I've tried reflashing but it doesn't help.
[betaflight_3.2.0_FRSKYF3X_170714 (1).hex.zip](https://github.com/betaflight/betaflight/files/1207596/betaflight_3.2.0_FRSKYF3X_170714.1.hex.zip)

Thanks again.

AndersHoglund, 2017-08-08:
[image]

applewood, 2017-08-08:
ok great, that's made them visible, thanks @AndersHoglund

applewood, 2017-08-08:
Yeah! It's working again.   Thanks @jflyper and @AndersHoglund .   Reflashing with the BF FRSKYF3 version of the firmware, I can now see the Gyro and OSD.     Must be an issue with the version I got sent from FRSKY.   

Thanks so much for your help, it's really appreciated.

jflyper, 2017-08-08:
Happy props :)

closed 2017-08-08 by jflyper