betaflight/betaflight #3776: PID and rate profiles are dumped only partially

andrejpodzimek, 2017-08-07:
Since a few releases back I noticed that two of the three PID profiles and rate profiles are not in the output of `dump`. I'm using a BlueJayF4 board now. The dump data only contains one of the profiles.

    # profile
    profile 0
    
    set dterm_lowpass_type = BIQUAD
    set dterm_lowpass = 100
    set dterm_notch_hz = 260
    set dterm_notch_cutoff = 160
    set vbat_pid_gain = ON
    set pid_at_min_throttle = ON
    set anti_gravity_threshold = 350
    set anti_gravity_gain = 2000
    set setpoint_relax_ratio = 100
    set dterm_setpoint_weight = 60
    set acc_limit_yaw = 100
    set acc_limit = 0
    set crash_dthreshold = 50
    set crash_gthreshold = 400
    set crash_setpoint_threshold = 350
    set crash_time = 500
    set crash_delay = 0
    set crash_recovery_angle = 10
    set crash_recovery_rate = 100
    set crash_recovery = OFF
    set iterm_windup = 50
    set yaw_lowpass = 90
    set pidsum_limit = 500
    set pidsum_limit_yaw = 400
    set p_pitch = 58
    set i_pitch = 50
    set d_pitch = 35
    set p_roll = 40
    set i_roll = 40
    set d_roll = 30
    set p_yaw = 70
    set i_yaw = 45
    set d_yaw = 20
    set p_alt = 50
    set i_alt = 0
    set d_alt = 0
    set p_level = 50
    set i_level = 50
    set d_level = 80
    set p_vel = 55
    set i_vel = 55
    set d_vel = 75
    set level_sensitivity = 55
    set level_limit = 55
    set horizon_tilt_effect = 75
    set horizon_tilt_expert_mode = OFF
    set gps_pos_p = 15
    set gps_pos_i = 0
    set gps_pos_d = 0
    set gps_posr_p = 34
    set gps_posr_i = 14
    set gps_posr_d = 53
    set gps_nav_p = 25
    set gps_nav_i = 33
    set gps_nav_d = 83
    
    # rateprofile
    rateprofile 0
    
    set rc_rate = 100
    set rc_rate_yaw = 100
    set rc_expo = 0
    set rc_expo_yaw = 0
    set thr_mid = 50
    set thr_expo = 0
    set roll_srate = 70
    set pitch_srate = 70
    set yaw_srate = 70
    set tpa_rate = 10
    set tpa_breakpoint = 1650

This^^^ is all the rate information I see in the dump. The other profiles are just not there. However, I did tweak and change all three of them, so they should be represented in the dump data somewhere.

wind0r, 2017-08-07:
use "dump all" if you want all profiles.

andrejpodzimek, 2017-08-07:
Thanks, I haven't known about `dump all`. Somehow I missed it, although there must have been a [long debate](https://github.com/betaflight/betaflight/issues/487) about this topic. So this is not a bug.

closed 2017-08-07 by andrejpodzimek