betaflight/betaflight #3777: No GPS fix and bogus position data with HGLRC UBLOX M8N

andrejpodzimek, 2017-08-07:
I'm trying to get a GPS fix on my drone with a [HGLRC UBLOX M8N](http://www.helipal.com/hglrc-ublox-gps-m8n-module.html) GPS sensor. This is a BlueJayF4 board running a recent master build. However, it just doesn't fix. My cell phone held right next to the drone gets ~9 satellites instantly, so it's not a matter of a weak signal or the like. This is the best reception I've ever got: 
[image]
This is insufficient for a GPS fix, presumably. The sensor should have a (cold) time to first fix of ~30 seconds, but it's still equally bad after an hour. I mentioned further details about my sensor placement and GPS settings [here](https://www.rcgroups.com/forums/showpost.php?p=38051727&postcount=18). At this point I'm making a wild guess that this may be a configuration issue. Some of these GPS sensors seem to have an option to set whether they are connected to a powered (pre-amplified) antenna system or not. Getting that wrong can cause bad signal reception. But I'm not sure whether it applies to this device. My current settings:

    gps_provider = UBLOX
    gps_sbas_mode = EGNOS
    gps_auto_config = ON
    gps_auto_baud = ON

I also tried to switch to `NMEA`, because the sensor should speak that protocol too. (Just in case if this was actually a wrong interpretation of the UBLOX data.) However, the GPS wasn't even recognized. Does `NMEA` require some special additional settings, such as one particular baud rate? (I thought the `gps_auto_baud` would do the trick.)

Also, the longitude and lattitude data I'm getting is utterly bogus. I'm in Zürich, which is nowhere near the indicated location. There could be indeed something wrong about the communication with the sensor.
[image]

Last but not least, I may have a faulty piece of hardware. :( Not sure how to find out though, because I don't have another similar sensor for comparison.

jflyper, 2017-08-07:
@andrejpodzimek 

- You can test your GPS module independent from BF using ublox's uCenter software.

andrejpodzimek, 2017-08-07:
I see. I'll give [this](http://www.mertl-research.at/ceonwiki/doku.php?id=software:u-center) a try then. But I don't have any USBserial converter suitable for that purpose. Can this work with `serialpassthrough` while connected to the FC? Also, it looks like the communication protocol involves a few baud rate changes, so should I start with e.g. `serialpassthrough 3 57600` or something else?

jflyper, 2017-08-07:
@andrejpodzimek 
- I think you can lock the baud rate to a specific value in u-Center,
- but you still have to find out the value by try and error with serialpassthrough and some terminal emulator software.

andrejpodzimek, 2017-08-07:
57600 works.
But the sensor doesn't work. I mean, the connection with u-Center works perfectly under Wine, but it shows the same total BS it showed before in Betaflight. I don't know what's wrong. uCenter can dump a diagnostic file, so I'll ask on the u-blox forums. This just doesn't look right at all:
[image]
Anyway, I think Betaflight is not to blame. Sorry for the noise.

closed 2017-08-07 by andrejpodzimek

andrejpodzimek, 2017-08-07:
[Posted a question here.](https://forum.u-blox.com/index.php?qa=12801) Well, now I'm really curious what's wrong with the device. Anyway, now I know that `serialpassthrough` simply rocks. It helped a lot in this case.

jflyper, 2017-08-07:
If it turned out to be a counterfeit, then it's the 2nd worst I've heard of.

andrejpodzimek, 2017-08-07:
That would be sad. Up to now I've been getting reasonable stuff from Helipal, nothing that would look like counterfeits of an inferior quality. But this particular device is indeed disappointing. I'm happy to invest time into looking for solutions and asking around, but broken hardware is definitely not worth anyone's time. :-/

andrejpodzimek, 2017-08-12:
Just for the record, I got a second one and [it was worse than the first one](https://forum.u-blox.com/index.php?qa=12801#12982). This is sad. I would be more than happy to pay u-blox $100 for a device that's guaranteed to work.

jflyper, 2017-08-12:
😭

andrejpodzimek, 2017-08-16:
[**This device works for me!**](https://www.banggood.com/NEO-M8N-Flight-Controller-GPS-Module-GYGPSV5-NEO-for-Pixhawk-APM-p-974362.html) It's slightly bigger, but it has a built-in I2C compass, HMC5983 (the one with temperature compensation), and it all just works fine out-of-the-box. On my balcony, which is covered nearly from all sides and only sees a small patch of sky, it got 6 satellites and a fix within 15 seconds. So. Much. Better.

So much for the good news.
I removed my separate I2C compass and started using the one built into the GPS instead. It's pretty much the same device, also on I2C. And I also hooked the I2C baro to it, the one that had worked up to now. But then I started getting quite a number of I2C errors and random baro jumps. [I report the details here.](https://github.com/betaflight/betaflight/issues/3861)