betaflight/betaflight #3781: NeroF7 - IRC Tramp can't be set on uart1

goodnerdboy, 2017-08-07:
repro:

jflyper, 2017-08-07:
@goodnerdboy 
(1) Did you do full-erase or cli `defaults`?
(2) What version of BFC are you using?

goodnerdboy, 2017-08-08:
@jflyper 
(1) i did full chip erase when flashing, and i did Reset Settings in the configurator.
(2) i am using 3.2.1 version of BFC

Changing the peripherals setting for uart1 in the port tab is the only modification i did for the repro steps.
I tried this on two different NeroF7.
One already mounted in a quad, and one completly new, not connected to anything.

Saxin, 2017-08-08:
@jflyper I don't have this particular FC but seems related to #3692

goodnerdboy, 2017-08-08:
@saxin I'll check tonight by resoldering to swap the uarts to check if it actualy works, but at least just setup wise, I can set IRC Tramp on any other uart in BFC without trouble but uart1.
Another thing is I fly build #1511 with irc tramp setup on uart1 and it works perfectly.

goodnerdboy, 2017-08-08:
So resoldered a little bit, now smartport on UART1 and IRC Tramp on UART6, i am able to put the right configuration in BFC, but setup the Tramp through my taranis doesn't work anymore, i can't neither read or right parameters on it.
Smartport and pid settings work in both configurations.
I'll fly with 3.2 build 1511, crossing fingers you find a fix :)
Tell me if i can test anything else.

jflyper, 2017-08-09:
@goodnerdboy I confirmed the problem with NERO and RC3, at least the loss of connectivity to configurator.

jflyper, 2017-08-16:
The loss of connectivity to BFC has been fixed by #3867.

goodnerdboy, 2017-08-16:
Awesome, flashed build #83 and it seems to work like a charm now.
IRC Tramp setup through betaflight works again. Great job, thanks!!!

jflyper, 2017-08-16:
Happy props :)

closed 2017-08-16 by jflyper